platformio/platform-atmelavr #2: Circuit Playground from Adafruit

bendo, 2016-06-11:
Hi, I try to use new board from Adafruit, Circuit Playground. It is based on Flora board. So in Platformio IDE I am creating project with flora board. Everything looks ok but program runs slowly. Because of this I've tried to create a new board (same as flora, just I changed name and speed to 9600.
But in IDE, when I try to upload there is an error like: 
```
scons: *** [.pioenvs/playground/src/tmp_ino_to.o] UserError `KeyError `'hwid'' trying to evaluate `${BOARD_OPTIONS[
'build']['hwid'][0][0]}'' trying to evaluate `${_defines(CPPDEFPREFIX, CPPDEFINES, CPPDEFSUFFIX, __env__)}'

```

Can you please help me. I try platformio IDE, and it looks great, just I have to solve this issue... thank you

ivankravets, 2016-06-11:
1. You don't need to create a new board. Just use [upload_speed](http://docs.platformio.org/en/latest/projectconf.html#upload-speed).
2. This repo is intended for PlatformIO 3.0 development platform
3. Do you use a development version of PlatformIO 2.0? You need to use `hwids` instead `hwid` in board JSON manifest.

bendo, 2016-06-11:
Ah, sorry. I saw repo pla, tform-atmelavr so I thought that here are all current boards. :-)
Hm, so my platformio.ini looks like this: 
```

[env:flora8]
platform = atmelavr
framework = arduino
board = flora8
upload_speed = 9600
```

I've tried upload_speed 9600 and 57600 but in both cases LED diodes blink really slow. Do you think issue is in the upload_speed? If yes, can you suggest some another speed which I can try. Btw: in demo.ini they opening Serial at 9600 but this speed also didn't work for me.

ivankravets, 2016-06-12:
Please don't mix UPLOAD_SERIAL_SPEED and MONITOR_SERIAL_SPEED. You don't need to change upload speed.

Do you have a problem when using Arduino IDE?

bendo, 2016-06-12:
Hm, so this playground circuit board has leds on it. The demo program, turns on leds one by one every second. When I upload program via Arduino IDE, it works as expected. If I upload same program with PlatformIO IDE, it turns on leds one by one but not every second but every cca 50 seconds.

ivankravets, 2016-06-23:
Did you resolve this issue?

closed 2016-08-03 by ivankravets

bendo, 2017-04-28:
Sorry for late response. But yes I've resolve the issue. Thanks for helping.