platformio/platform-atmelavr #3: Flash is not erased when uploading hexfile

artur-scholz, 2016-07-14:
Using the Platformio IDE integrated in Atom, when performing "Upload" the command line output shows that the "-D" flags is used with the avrdude tool. 

This disables auto erase for flash. However, it is essential that the flash is erased before uploading new code into it. I am using a bare Atmega48 (no Arduino). I did not find a way to remove this flag anywhere in the platformio config files..

ivankravets, 2016-07-14:
Use `program` instead `upload`. See http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

`program` is bare target and you can customize it manually using `upload_flags`.

artur-scholz, 2016-07-14:
Great! Is it possible to have this integrated in the left toolbar? Otherwise I need to select "PlatformIO: Upload using programmer" either via left bottom option or via F7.

ivankravets, 2016-07-14:
Em.... Very interesting question 👍 There are a lot of solutions to do it 😄 

How about to create [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script) and override with custom upload command?

See my example here https://github.com/platformio/platformio/issues/247#issuecomment-118169728

artur-scholz, 2016-07-14:
Thanks for the fast reply. Honestly I want to avoid tinkering with scripts. So I will use the available option then.

closed 2016-07-14 by artur-scholz

ivankravets, 2016-07-14:
Another option is to use [targets](http://docs.platformio.org/en/latest/projectconf.html#targets). In this case, "Build" button will have "program" behaviour.

For example,
```ini
[env:my]
targets = program
```