platformio/platform-atmelavr #5: usbdrvasm.asm issue on Digispark

kickscondor, 2016-08-12:
Hey there - thanks for the work on AVRs for PlatformIO. It's great stuff! My issue is fortunately very minor - but seems to be buried in a tarball somewhere.

After a fresh install of PlatformIO, I run into an issue building projects that use DigisparkCDC. The error is:

```
scons: *** Multiple ways to build the same target were specified for: /Users/kicks/Code/astrochip/spark/.pioenvs/digispark-pro/DigisparkCDC/usbdrvasm.o  (from ['/Users/kicks/Code/astrochip/spark/.pioenvs/digispark-pro/DigisparkCDC/usbdrvasm.S'] and from ['/Users/kicks/Code/astrochip/spark/.pioenvs/digispark-pro/DigisparkCDC/usbdrvasm.asm'])
File "/usr/local/Cellar/platformio/2.11.2/libexec/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 257, in BuildLibrary
```

I experience this on OSX and Linux alike. I can circumvent this issue like so:

```
cd ~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkCDC
mv usbdrvasm.asm usbdrvasm.asm~
```

That's all it takes. Many thanks!

valeros, 2016-09-02:
Sorry for the late reply. `framework-arduinoavr` package has been updated.
Please run `platformio update`

closed 2016-09-02 by valeros

kickscondor, 2016-09-09:
Hey, looks good. Thanks a lot!