platformio/platform-atmelavr #7: Update to Atmel AVR toolchain

zgoda, 2016-08-19:
Arduino 1.6.10 brings some build optimizations in toolchain resulting in smaller firmware with Atmel release 3.5.3. There was update to avrdude in this release but later in 1.6.11 avrdude has been reverted to 6.0.1 because of regressions.

ivankravets, 2016-09-08:
Related topic https://community.platformio.org/t/support-for-atmega328pb/314

zgoda, 2016-09-29:
1.6.12 finally updates avrdude to 6.3.

ekarlso, 2016-10-30:
Is there any news on why this havent been done ? I want to use the Atmega328PB which alledgedly is supported by updating this.

ivankravets, 2016-10-30:
Could someone help us with new packages? We will replace old that are located here
https://bintray.com/platformio/dl-packages/toolchain-atmelavr#files

ekarlso, 2016-10-30:
What do you mean by help you ? Download the packages from Atmel for you ?

ivankravets, 2016-10-31:
> Download the packages from Atmel for you ?

Good joke!

-------

I mean that multiple developers migrate from Arduino IDE to PlatformIO IDE. We should be confident that new users will not have problems with projects that work with Arduino IDE and don't work with PlatformIO IDE because we use different toolchains, etc.

The question: does Arduino IDE uses Atmel AVR 3.5.3 toolchain?

ekarlso, 2016-10-31:
Q then, is it possible to maybe define a custom toolchain for a particulat project or similar since I have the atmega328 ?

ivankravets, 2016-10-31:
Yes, it is. You can clone existing Atmel Avr development platform and specify own toolchain version.

In any case, I am ready to update generic GCC-AVR toolchain but need to be confident that we will not break thousands of PlatformIO projects over the world.

Could someone check this information? Which version of gcc-avr is used in Arduino IDE? Then, we will decide together here which we should use in PlatformIO.

ekarlso, 2016-10-31:
The newest Arduino uses 4.9.2
ekarlso@ekarlso-tpad:~/Downloads/arduino-1.6.12$ ./hardware/tools/avr/bin/avr-g++
avr-g++: fatal error: no input files
compilation terminated.
ekarlso@ekarlso-tpad:~/Downloads/arduino-1.6.12$ ./hardware/tools/avr/bin/avr-g++ -v
Using built-in specs.
Reading specs from /home/ekarlso/Downloads/arduino-1.6.12/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/device-specs/specs-avr2
COLLECT_GCC=./hardware/tools/avr/bin/avr-g++
COLLECT_LTO_WRAPPER=/home/ekarlso/Downloads/arduino-1.6.12/hardware/tools/avr/bin/../libexec/gcc/avr/4.9.2/lto-wrapper
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/workspace/avr-gcc/label/debian7-x86_64/objdir --enable-long-long --disable-nls --disable-checking --disable-libssp --disable-libada --disable-shared --enable-lto --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 4.9.2 (GCC)

ivankravets, 2016-10-31:
@ekarlso thanks! Is this toolchain the same that we can downloads from Atmel site?

ekarlso, 2016-10-31:
That is the bundled version of Arduino 1.6.12.

This is the avr toolchain 3.5.4
Using built-in specs.
Reading specs from /home/ekarlso/Downloads/avr8-gnu-toolchain-linux_x86_64/bin/../lib/gcc/avr/4.9.2/device-specs/specs-avr2
COLLECT_GCC=./bin/avr-g++
COLLECT_LTO_WRAPPER=/home/ekarlso/Downloads/avr8-gnu-toolchain-linux_x86_64/bin/../libexec/gcc/avr/4.9.2/lto-wrapper
Target: avr
Configured with: /home/toolsbuild/workspace/avr8-gnu-toolchain/src/gcc/configure LDFLAGS=-L/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64-hostlibs/lib CPPFLAGS= --target=avr --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64 --libdir=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64/lib --enable-languages=c,c++ --with-dwarf2 --enable-doc --disable-shared --disable-libada --disable-libssp --disable-nls --with-avrlibc=yes --with-mpfr=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64-hostlibs --with-gmp=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64-hostlibs --with-mpc=/home/toolsbuild/workspace/avr8-gnu-toolchain/avr8-gnu-toolchain-linux_x86_64-hostlibs --with-pkgversion=AVR_8_bit_GNU_Toolchain_3.5.4_1709 --with-bugurl=http://www.atmel.com
Thread model: single
gcc version 4.9.2 (AVR_8_bit_GNU_Toolchain_3.5.4_1709)

ekarlso, 2016-10-31:
[Mon Oct 31 14:31:34 2016] Processing controller (upload_protocol: usbasp, framework: arduino, lib_install: 13,135, upload_flags: -Pusb -B5, platform: atmelavr, board: gatekeeper-controller)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 30 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <rs485-nodeproto>
Compiling .pioenvs/controller/src/main.o
In file included from /home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:90:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:28,
from src/main.cpp:1:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
#    warning "device type not defined"
^
Compiling .pioenvs/controller/FrameworkArduino/CDC.o
In file included from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:25:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:225,
from src/main.cpp:1:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
# warning "Device does not have EEPROM available."
^
Compiling .pioenvs/controller/FrameworkArduino/HardwareSerial.o
In file included from /home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:90:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:24,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/CDC.cpp:19:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
#    warning "device type not defined"
^
In file included from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:25:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/CDC.cpp:19:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
# warning "Device does not have EEPROM available."
^
Compiling .pioenvs/controller/FrameworkArduino/HardwareSerial0.o
src/main.cpp:6:27: error: 'Serial' was not declared in this scope
RS485Controller rs485ctrl(Serial, RS485_RE, RS485_WE);
^
src/main.cpp: In function 'void setup()':
src/main.cpp:18:3: error: 'Serial' was not declared in this scope
Serial.begin(9600);
^
src/main.cpp: In function 'void loop()':
src/main.cpp:33:7: error: 'Serial' was not declared in this scope
Serial.println("RX: IDLE");
^
*** [.pioenvs/controller/src/main.o] Error 1
In file included from /home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:90:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:28,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial.cpp:29:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
#    warning "device type not defined"
^
In file included from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:25:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:225,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial.cpp:29:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
# warning "Device does not have EEPROM available."
^
In file included from /home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:90:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:28,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial0.cpp:25:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
#    warning "device type not defined"
^
In file included from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:25:0,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:225,
from /home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial0.cpp:25:
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
# warning "Device does not have EEPROM available."
^
[... 13 lines cut ...]
^
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/wdt.h:482:24: error: 'WDCE' was not declared in this scope
"r" ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE))),
^
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/wdt.h:482:42: error: 'WDE' was not declared in this scope
"r" ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE))),
^
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/wdt.h: In function 'void wdt_disable()':
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/wdt.h:494:21: error: 'WDT' was not declared in this scope
[... cut ...]
[WDCE_WDE]  "n"  ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE)))
^
/home/ekarlso/.platformio/packages/toolchain-atmelavr/avr/include/avr/wdt.h:526:55: error: 'WDE' was not declared in this scope
[WDCE_WDE]  "n"  ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE)))
^
*** [.pioenvs/controller/FrameworkArduino/CDC.o] Error 1
========================================================================================================= [ERROR] Took 0.64 seconds =========================================================================================================

ivankravets, 2016-10-31:
What is it? New toolchain?

ekarlso, 2016-10-31:
That is the Atmel 3.5.4 toolchain (I just swapped toolchain-atmelavr with this one)

ivankravets, 2016-10-31:
Could you try Arduino IDE's toolchain?

Also, check that new toolchain structure is the same as previous and all symbolic links are resolved.

ekarlso, 2016-10-31:
So I did the following:
cp -R /home/ekarlso/arduino-1.6.12/hardware/arduino/avr/ framework-arduinoavr
cp -R /home/ekarlso/arduino-1.6.12/hardware/tools/avr/ toolchain-atmelavr

copied package.json from the old versions to the "copy" and then tried to run "pio run" in my project but it fails giving
*** Multiple ways to build the same target were specified for: /home/ekarlso/projects/gatekeeper/gatekeeper-rs485-reader/.pioenvs/reader/FrameworkArduino/wiring_pulse.o  (from ['/home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/wiring_pulse.S'] and from ['/home/ekarlso/.platformio/packages/framework-arduinoavr/cores/arduino/wiring_pulse.c'])
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 263, in BuildLibrary

So I checked the "old" version and I see you have renamed the file to _file.S there and that fixed it and my project compiles.

[arduino_1.6.12.txt](https://github.com/platformio/platform-atmelavr/files/561871/arduino_1.6.12.txt)
[arduino-old.txt](https://github.com/platformio/platform-atmelavr/files/561873/arduino-old.txt)

ivankravets, 2016-10-31:
Stop... We talk here about toolchain but not about new Arduino Framework.

Need to check Arduino IDE's toolchain-atmelavr

ivankravets, 2016-10-31:
> cp -R /home/ekarlso/arduino-1.6.12/hardware/tools/avr/ toolchain-atmelavr

Please sorry, I see it.

Does it work with  Atmega328PB ?

ekarlso, 2016-10-31:
You need to add support for it by the looks by copying the files in before pio run is done.

I have updated the project so you can see what is being done
https://gitlab.com/creator-makerspace/gatekeeper-rs485-controller/blob/develop/bootstrap.sh

basically it's a very hacky solution atm to get it going.
1. "pio run" to get the toolchain etc installed
2. replace the packages/toolchain-avr with the toolchain from arduino 1.6.12
3. bootstrap.sh to install support for the atmega328pb
4. run platformio run to compile..

And after all that there's still no support in the current avrdude.conf that is with platformio for it.

ekarlso, 2016-10-31:
Any idea @ivankravets ?

ivankravets, 2016-10-31:
I'll update all toolchains and add support for atmega328pb in PlatformIO 3.2.

ivankravets, 2016-10-31:
The `boards_dir` is already implemented https://github.com/platformio/platformio/commit/7b4d94ccfb7c5ddf55d87338a673d226265a2a78

ekarlso, 2016-10-31:
If you want me to test dev version I can do that once you update ;)

ivankravets, 2016-10-31:
Sure, I'll ask here. I'm working on the first public beta of PlatformIO 3.2 and after it will take a look at the new packages, boards, etc.

Would you like to try out PIO Remote? http://docs.platformio.org/en/latest/plus/pio-remote.html

Would be thankful for any feedback!

ivankravets, 2016-11-18:
Avrdude has been updated to 6.3. Also , added support for ATmega328PB

ivankravets, 2016-11-19:
@ekarlso Could you help us to resolve this issue? https://ci.appveyor.com/project/ivankravets/platform-atmelavr/build/1.0.101/job/8m3his6x75rs9bo8#L2284

I patched all toolchains with https://github.com/watterott/ATmega328PB-Testing but it doesn't work with Windows toolchain.

ivankravets, 2017-11-27:
Fixed toolchain for Windows

commit 25381af1 (2016-11-16): Update Atmel AVR toolchain to GCC 4.9.2/Atmel 3.5.4 // Resolve #7
closed 2017-11-27 by ivankravets