platformio/platform-atmelavr #10: avrdude can't handle file paths with spaces in it anymore

KarateBrot, 2016-09-10:
[09/10/16 14:02:38] Processing uno (platform: atmelavr, board: uno, framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ClickEncoder>

|-- <Adafruit MCP4725> v1.0.2
|   |-- <Wire> v1.0
|-- <Wire> v1.0
|-- <Adafruit SSD1306> v1.1.0
|   |-- <Adafruit GFX Library> v1.1.5
|   |-- <SPI> v1.0
|   |-- <Wire> v1.0
|-- <Adafruit GFX Library> v1.1.5
|-- <TimerOne>
|-- <Adafruit INA219> v1.0.0
|   |-- <Wire> v1.0
Linking .pioenvs\uno\firmware.elf
Checking program size .pioenvs\uno\firmware.elf
text       data     bss     dec     hex filename
19294      1182     567   21043    5233 .pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
Looking for upload port...
Auto-detected: COM15
Uploading .pioenvs\uno\firmware.hex
avrdude: can't open config file "C:\Users\Jan": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\Jan"
*** [upload] Error 1
 [ERROR] Took 1.13 seconds

- - - - -

After updating to PlatformIO 3.0.1 avrdude seems to cut away the rest of the file path for the config file. 
This bug already popped up two days ago when I enabled development version 3.0.0b13.
This was not an issue in the last stable release before 3.0.1.

ivankravets, 2016-09-10:
What is your username in system?

KarateBrot, 2016-09-10:
Jan Post

The file path stops after "Jan" right before the space

ivankravets, 2016-09-10:
Please open PlatformIO IDE Terminal and run `pio run -v`. Share here results. Let' fix it. Thanks.

KarateBrot, 2016-09-10:
[09/10/16 14:27:09] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph

|-- <Adafruit INA219> v1.0.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_INA219)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit MCP4725> v1.0.2 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_MCP4725)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|-- <ClickEncoder> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\ClickEncoder)
|-- <TimerOne> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\TimerOne)
|-- <Adafruit SSD1306> v1.1.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_SSD1306)
|   |-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|   |-- <SPI> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\SPI)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
avr-g++ -o .pioenvs\uno\firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--relax .pioenvs\uno\src\Vaporizer.o -L.pioenvs\uno -Wl,--start-group .pioenvs\uno\libFrameworkArduinoVariant.a .pioenvs\uno\libFrameworkA
rduino.a -lm .pioenvs\uno\lib\libWire.a .pioenvs\uno\lib\libAdafruit_INA219.a .pioenvs\uno\lib\libAdafruit_MCP4725.a .pioenvs\uno\lib\libAdafruit_GFX.a .pioenvs\uno\lib\libClickEncoder.a .pioenvs\uno\lib\libTimerOn
e.a .pioenvs\uno\lib\libSPI.a .pioenvs\uno\lib\libAdafruit_SSD1306.a -Wl,--end-group
avr-objcopy -O ihex -R .eeprom .pioenvs\uno\firmware.elf .pioenvs\uno\firmware.hex
avr-size --mcu=atmega328p -C -d .pioenvs\uno\firmware.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:   20498 bytes (62.6% Full)
(.text + .data + .bootloader)

Data:       1749 bytes (85.4% Full)
(.data + .bss + .noinit)

============================================================================================ [SUCCESS] Took 0.86 seconds ============================================================================================

ivankravets, 2016-09-10:
Sorry, please run `pio run -t upload`

KarateBrot, 2016-09-10:
[09/10/16 14:36:52] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 34 compatible libraries

Looking for dependencies...
Library Dependency Graph
|-- <Adafruit SSD1306> v1.1.0
|   |-- <Adafruit GFX Library> v1.1.5
|   |-- <SPI> v1.0
|   |-- <Wire> v1.0
|-- <Wire> v1.0
|-- <Adafruit INA219> v1.0.0
|   |-- <Wire> v1.0
|-- <Adafruit GFX Library> v1.1.5
|-- <Adafruit MCP4725> v1.0.2
|   |-- <Wire> v1.0
|-- <ClickEncoder>
|-- <TimerOne>
Linking .pioenvs\uno\firmware.elf
Checking program size .pioenvs\uno\firmware.elf
text       data     bss     dec     hex filename
19314      1182     567   21063    5247 .pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
Looking for upload port...
Auto-detected: COM15
Uploading .pioenvs\uno\firmware.hex
avrdude: can't open config file "C:\Users\Jan": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\Jan"
*** [upload] Error 1
============================================================================================= [ERROR] Took 1.10 seconds =============================================================================================

ivankravets, 2016-09-10:
Eh :( Sorry, with verbose, `pio run -t upload -v`. Need to see full command.

KarateBrot, 2016-09-10:
Sorry, my bad...

[09/10/16 14:40:28] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit INA219> v1.0.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_INA219)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)

|-- <Adafruit MCP4725> v1.0.2 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_MCP4725)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|-- <ClickEncoder> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\ClickEncoder)
|-- <TimerOne> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\TimerOne)
|-- <Adafruit SSD1306> v1.1.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_SSD1306)
|   |-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|   |-- <SPI> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\SPI)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
avr-g++ -o .pioenvs\uno\firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--relax .pioenvs\uno\src\Vaporizer.o -L.pioenvs\uno -Wl,--start-group .pioenvs\uno\libFrameworkArduinoVariant.a .pioenvs\uno\libFrameworkA
rduino.a -lm .pioenvs\uno\lib\libWire.a .pioenvs\uno\lib\libAdafruit_INA219.a .pioenvs\uno\lib\libAdafruit_MCP4725.a .pioenvs\uno\lib\libAdafruit_GFX.a .pioenvs\uno\lib\libClickEncoder.a .pioenvs\uno\lib\libTimerOn
e.a .pioenvs\uno\lib\libSPI.a .pioenvs\uno\lib\libAdafruit_SSD1306.a -Wl,--end-group
Checking program size .pioenvs\uno\firmware.elf
text       data     bss     dec     hex filename
19316      1182     567   21065    5249 .pioenvs\uno\firmware.elf
avr-objcopy -O ihex -R .eeprom .pioenvs\uno\firmware.elf .pioenvs\uno\firmware.hex
BeforeUpload(["upload"], [".pioenvs\uno\firmware.hex"])
Auto-detected: COM15
avrdude -v -p atmega328p -C ""C:\Users\Jan Post\.platformio\packages\tool-avrdude\avrdude.conf"" -c arduino -b 115200 -P "COM15" -D -U flash:w:.pioenvs\uno\firmware.hex:i

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "C:\Users\Jan"
avrdude: can't open config file "C:\Users\Jan": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\Jan"
*** [upload] Error 1
============================================================================================= [ERROR] Took 1.08 seconds =============================================================================================

ivankravets, 2016-09-10:
Thanks, looks like a bug. I'll try to reproduce it in my virtual machine

ivankravets, 2016-09-10:
Fixed in https://github.com/platformio/platform-atmelavr/commit/59b6ab1691821f4679ab4625c72a7ec01375830d

Could you test it?

1. Open PlatformIO IDE Terminal
2. Uninstall current version of development platform via 
```
pio platform uninstall atmelavr
```
3. Install fixed development version:
```
pio platform install https://github.com/platformio/platform-atmelavr/archive/develop.zip
```

If it works, I'll make release.

KarateBrot, 2016-09-10:
It works again!
I pasted the results of the terminal message in here again, just in case you want to have a look:

- - - - -

[09/10/16 19:19:10] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit MCP4725> v1.0.2 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_MCP4725)

|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit INA219> v1.0.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_INA219)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|-- <ClickEncoder> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\ClickEncoder)
|-- <TimerOne> (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\TimerOne)
|-- <Adafruit SSD1306> v1.1.0 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_SSD1306)
|   |-- <Adafruit GFX Library> v1.1.5 (D:\Users\Jan Post\Documents\Arduino\Vaporizer\lib\Adafruit_GFX)
|   |-- <SPI> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\SPI)
|   |-- <Wire> v1.0 (C:\Users\Jan Post\.platformio\packages\framework-arduinoavr\libraries\Wire)
avr-g++ -o .pioenvs\uno\firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--relax .pioenvs\uno\src\Vaporizer.o -L.pioenvs\uno -Wl,--start-group .pioenvs\uno\libFrameworkArduinoVariant.a .pioenvs\uno\libFrameworkA
rduino.a -lm .pioenvs\uno\lib\libWire.a .pioenvs\uno\lib\libAdafruit_MCP4725.a .pioenvs\uno\lib\libAdafruit_INA219.a .pioenvs\uno\lib\libAdafruit_GFX.a .pioenvs\uno\lib\libClickEncoder.a .pioenvs\uno\lib\libTimerOn
e.a .pioenvs\uno\lib\libSPI.a .pioenvs\uno\lib\libAdafruit_SSD1306.a -Wl,--end-group
Checking program size .pioenvs\uno\firmware.elf
text       data     bss     dec     hex filename
19316      1182     567   21065    5249 .pioenvs\uno\firmware.elf
avr-objcopy -O ihex -R .eeprom .pioenvs\uno\firmware.elf .pioenvs\uno\firmware.hex
BeforeUpload(["upload"], [".pioenvs\uno\firmware.hex"])
Auto-detected: COM15
avrdude -v -p atmega328p -C "C:\Users\Jan Post\.platformio\packages\tool-avrdude\avrdude.conf" -c arduino -b 115200 -P "COM15" -D -U flash:w:.pioenvs\uno\firmware.hex:i

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "C:\Users\Jan Post\.platformio\packages\tool-avrdude\avrdude.conf"

Using Port                    : COM15
Using Programmer              : arduino
Overriding Baud Rate          : 115200
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated

RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : Arduino
Description     : Arduino
Hardware Version: 3
Firmware Version: 4.4
Vtarget         : 0.3 V
Varef           : 0.3 V
Oscillator      : 28.800 kHz
SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ".pioenvs\uno\firmware.hex"
avrdude: writing flash (20498 bytes):

Writing | ################################################## | 100% 3.30s

avrdude: 20498 bytes of flash written
avrdude: verifying flash memory against .pioenvs\uno\firmware.hex:
avrdude: load data flash data from input file .pioenvs\uno\firmware.hex:
avrdude: input file .pioenvs\uno\firmware.hex contains 20498 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.64s

avrdude: verifying ...
avrdude: 20498 bytes of flash verified

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.

============================================================================================ [SUCCESS] Took 9.28 seconds ============================================================================================

ivankravets, 2016-09-10:
Thanks! I've just released a new version of Atmel AVR dev platform.

Run in IDE `Menu: PlatformIO > Update plaforms/packages/libs`. The platformio should be updated to 1.0.2

Thanks for the report!

closed 2016-09-10 by ivankravets