platformio/platform-atmelavr #11: Build errors with digispark board

igsol, 2016-09-12:
# Configuration
OS: Ubuntu 16.04 x64
Platformio version: 3.0.1

# Bug description
The brand new project, simplest Blink example. The 'platformio run' command leads error and warnings.

# Steps to reproduce

1. Create new directory (e.g. delme02) and cd into it.
2. Then execute the commands like in following session:
```bash
└─(12:15 $)─> platformio boards digispark

Platform: atmelavr
------------------------------------------------------------------------------------------------------------------------------------------------
ID                    MCU            Frequency  Flash   RAM    Name
------------------------------------------------------------------------------------------------------------------------------------------------
digispark-pro         ATTINY167      16Mhz     14kB    512B   Digistump Digispark Pro (Default 16 MHz)
digispark-pro32       ATTINY167      16Mhz     14kB    512B   Digistump Digispark Pro (16 MHz) (32 byte buffer)
digispark-pro64       ATTINY167      16Mhz     14kB    512B   Digistump Digispark Pro (16 MHz) (64 byte buffer)
digispark-tiny        ATTINY85       16Mhz     5kB     512B   Digistump Digispark (Default - 16 MHz)

└─(12:15 $)─> platformio init --board digispark-tiny

The current working directory /home/igor/prj/embed/delme02 will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /home/igor/prj/embed/delme02
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information

└─(12:16 $)─> platformio update
Platform Manager
================
Platform atmelavr
--------
Updating atmelavr @ 1.0.2: 	[Up-to-date]
Updating framework-arduinoavr @ 1.10608.2: 	[Up-to-date]
Updating toolchain-atmelavr @ 1.40801.0: 	[Up-to-date]
Updating tool-scons @ 2.4.1: 	[Up-to-date]

Library Manager
===============

└─(12:17 $)─> cat src/Blink.cpp 
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
[... 120 lines cut ...]
/home/igor/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
#define BIN     7
^
In file included from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Stream.h:24:0,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/TinyDebugSerial.h:31,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/TinyDebugSerial9600.cpp:24:
/home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Print.h:37:0: warning: "BIN" redefined [enabled by default]
#define BIN 2
^
In file included from /home/igor/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotn85.h:38:0,
from /home/igor/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:428,
from /home/igor/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:88,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Print.h:30,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Stream.h:24,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/TinyDebugSerial.h:31,
from /home/igor/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/TinyDebugSerial9600.cpp:24:
/home/igor/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
#define BIN     7
^
========================================================== [ERROR] Took 0.57 seconds ==========================================================
```
# Observed results
Error and BIN redefinition.
# Expected results
No errors and successfully built firmware :)

igsol, 2016-10-07:
Any suggestions how to overcome the aforementioned compilation error?

igsol, 2016-10-19:
LED_BUILTIN isn't applicable in case of digispark-tiny => the bug isn't valid.

closed 2016-10-19 by igsol

tcurdt, 2018-11-05:
Then use `platformio run --target upload` and it should work.