platformio/platform-atmelavr #12: add support for stk-600 and new tiny and mega devices

syddo, 2016-09-27:
Hi,

i'd like to add support for the Atmel STK-600 and the new ATMega324PB, ATtiny817 devices also.
how do i add a framework for the native avr-gcc?

ivankravets, 2016-09-27:
Do we need special toolchain for these boards?

syddo, 2016-10-06:
No. But the current avr-gcc needs to be updated to the latest release from Atmel.

ivankravets, 2016-10-09:
Ok, please help us.

The AVR toolchain is located in `~/.platformio/toolchain-atmelavr`. You can replace it with own version. Please keep `package.json` file without changes in that place. Otherwise, PlatformIO Package Manager will not see this package.

If it works, we will update current toolchain with your version. See https://bintray.com/platformio/dl-packages/toolchain-atmelavr#files