platformio/platform-atmelavr #14: Verify upload to AtTiny85 fails.

NormanDunbar, 2016-10-05:
ORIGINALLY RAISED WITH PLATFORMIO. ADVISED TO TAKE IT HERE. 

### Configuration

**Operating system**: Linux Mint 18, 64 bit.

**PlatformIO Version**: IDE=1.6.0, CLI=3.1.0

### Description of problem
After upgrading to latest PlatformIO, all my AtTiny85 scripts fail to verify. Each is Arduino compatible, and work fine on the Arduino IDE.

Please refer back to https://github.com/platformio/platformio/issues/799 and https://github.com/platformio/platformio/issues/800 for the full details.

I upgraded the Atom packages to the latest versions, now I cannot upload to an AtTiny85 using my USBTinyISP programmer any more. If failsattheverigy stage of `avrdude`. 

I followed the instaructions for the use of a programmer by visiting http://docs.platformio.org/en/stable/platforms/atmelavr.html#upload-using-programmer and my current platformio.ini is as follows:

**The content of `platformio.ini`:**
```ini
[env:attiny85]
platform=atmelavr
board=attiny85
framework=arduino
upload_protocol = usbtiny

[platformio]
src_dir=AtTiny85_Blink
```

After adding the upload_protocol option, _exactly_ the same happens. Same error message - unable to verify after uploading. I'm advised that I need to be using g program rather than upload,  but it doesn't seem to change anything. I'm still seeing g, briefly, the words platformio and upload as opposed to program.

With the Arduino IDE, I don't even have to "upload using programmer" to get a valid write and verify, upload works just fine. 

With PlatformIO, until I upgraded, I had no problems uploading - as opposed to programming - with the platformio.ini that I originally posted. Only after the upgrade were there any problems. I'm still at the point of being unable to program any of my AtTiny85 chips.

Changing to the advised upload_protocol had no effect. Still fails to verify the upload.

Sorry. :(

Cheers.

stonehippo, 2016-10-30:
I can confirm that I am having the same issu e with multiple ATTiny85s. I've tried using both a Sparkfun Tiny USB programmer and an Adafruit Trinket 3v. Both fail to verify after upload. Can upload fine from Arduino IDE.

StevenChristy, 2016-11-18:
Same issue. Using PlatformIO in Atom all 3 of my attiny85-20pu chips could be programmed once using a brand new Sparkfun Tiny AVR Programmer. Any subsequent changes to the program are made, I upload and the verification fails. I found that if I used the Arduino IDE function "Burn bootloader" on the chip then it could be reprogrammed again, exactly once. No verify failure is returned if you keep uploading the same program repeatedly. Could it be that the flash is not getting erased properly?

StevenChristy, 2016-11-18:
Nevermind, I found the resolution to my problem. I need to "program" instead of "upload" - http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

StevenChristy, 2016-11-18:
Just thought I'd share some more info on this. I found it annoying to use "program" every time as I am already used to hitting the upload button in the IDE. A simple fix for this was to tell avrdude to erase the flash before writing by adding the -e option in the upload_flags variable in the platformio.ini file. So the basic config that solves the problem is:

```
[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
upload_protocol = usbtiny
upload_flags = -e
```

ivankravets, 2016-11-18:
@NormanDunbar, @stonehippo could you confirm that solutions also work for you?

* https://github.com/platformio/platform-atmelavr/issues/14#issuecomment-261449607
* https://github.com/platformio/platform-atmelavr/issues/14#issuecomment-261455680

NormanDunbar, 2016-11-18:
Evening All,

I can confirm that changing from upload to program had no effect - I'd already tried that way back at the start of this problem.

I can also confirm that adding the "upload_flags = -e" option does work perfectly. I notice that the Arduino software does this anyway, so there's no problem in running an erase first.

Many thanks to all for this workaround. I appreciate the hard work, time and effort you all put in. Thanks.

Cheers,
Norm.

stonehippo, 2016-11-19:
@ivankravets Yes, using `upload_flags = -e` resolves the issue. I'm able to properly upload to via the AVR Tiny USB Programmer and to an Adafruit Trinket 3V with that flag in place in.

NormanDunbar, 2016-12-07:
Evening All,

I'm hpy that the workaround works and has resolved my problem. If the development team are happy too, we can close this issue.

Happy Christmas one and all, which applies to whichever religious, or otherwise, festival you celebrate at this time of year. ;-)

Cheers,
Norm.

ivankravets, 2016-12-07:
This is a bug. I've just fixed it. Please remove `upload_flags = -e` line from your config and run `pio update` command. All should work.

ivankravets, 2016-12-07:
@NormanDunbar 

Thanks! Merry Christmas! 🎄

closed 2016-12-07 by ivankravets

stonehippo, 2016-12-08:
@ivankravets look like it's working well. After `pio update`, I was able to program the ATTiny85 multiple times via the Tiny AVR Programmer, using both `upload` and `program` targets, with or without `upload_protocol = usbtiny` specified in `platformio.ini`. (At least I was after I remembered that it is flaky with my USB hub and plugged directly into my Mac!)

commit 3f9ce715 (2016-12-07): Handle extra upload flags from the boards // Resolve #14