platformio/platform-atmelavr #21: Add support for AVR XMEGA

ivankravets, 2016-11-27:
# Moved from https://github.com/platformio/platformio/issues/481

/cc @knutel, @aethaniel 

----------

It would be great to have support for AVR XMEGA, too. There already is a project to make Arduino support at [https://github.com/Xmegaduino/Xmegaduino], so maybe PlatformIO support is not that difficult. I might take a stab at it myself if I get the time.

corymayer, 2017-11-26:
Instead of porting the Arduino libraries would it be possible to use the standalone Atmel Software Framework? It's currently compatible with the XMEGA family and builds with avr-gcc.

http://www.atmel.com/tools/avrsoftwareframework.aspx

DaSimme, 2019-07-01:
Any progress on this? 

There is the project for Arduino IDE: https://github.com/XMegaForArduino/arduino
But would be cool to just add it to platform.io

she11sh0cked, 2020-11-16:
Are there still any plans to add support for the XMEGA boards?