platformio/platform-atmelavr #23: Programming Arduino Yun over wifi?

ivankravets, 2016-12-04:
# Moved from https://github.com/platformio/platformio-core/issues/452

/cc @grzegorzshub, @AKBON, @yveshanoulle

-----------------

Is it possible to program Yun over wifi please?

yveshanoulle, 2016-12-04:
what does this move mean? Is this being worked on?

ivankravets, 2016-12-04:
@yveshanoulle we cleanup PlatformIO Core repository from non-related issues. According this issue it will be automatically resolved with [PIO Remote](http://docs.platformio.org/en/latest/plus/pio-remote.html). PIO Remote allows to work with hardware from anywhere in the world. No need to open SSH or etc.

yveshanoulle, 2016-12-04:
Not really a solution for me. My devices are running on a private network. 
I don't want them to connect to the internet.

yveshanoulle, 2016-12-05:
+ I don't see arduino listed as supported platform. 
At this moment I have already to rewrite some of my code to make sure I can load everything on my arduino yun's as they have limited capacity. I assume that adding PIO Remote would make things worse (and a connection to the internet, will make them more vulnarable...
so I hope you really do support the same functionality that the default arduino editor supports without extra functionality that is not needed.

ivankravets, 2016-12-05:
@yveshanoulle PlatformIO Core doesn't have any relations to Arduino or other vendors. It is multi-platform, cross-platform and independent ecosystem. The other vendor related things are implemented via "development platform". This is a place where we have discussion now.

PlatformIO is free and open source ecossytem licensed under Apache 2.0. Anyone can improve this Atmel AVR development platform and make PR.

We are focused on the core features for PlatformIO that could be interested for thousands embedded developers not only for Arduino Yun owners.

Here is a docs:

* http://docs.platformio.org/en/latest/projectconf.html#extra-script
* http://docs.platformio.org/en/latest/platforms/creating_platform.html

If we willl have extra free time or a grant for this issue, [The PlatformIO Plus Team](https://pioplus.com/contact.html) will implement this issue.

yveshanoulle, 2016-12-05:
I understand it's unreleated to any vendor or platform, yet it's also "marketed" as a platform where we can program for arduino . and yes it's a much better editor then the basic arduino editor. I'm happy with that, only you don't offer the features that are needed to work in a production environment with the arduino's. Connecting with usb is not practical in any other setting then a testing or a play environment. In a coderdojo with the kids that is exactly what we do and that is perfect. 
I understand that for you it's not part of the platform core, yet I want to make sure you understand that the platform remote is not the same as what was existing in the basic arduino environment. It is much more complex yet forcing people to work in an environment with internet and for devices where memory is not scarse.

ivankravets, 2016-12-05:
1. I agree that PIO Remote != the same solution for this issue
2. I don't force people to work in an environment with internet or even to use PlatformIO
3. I proposed PIO Remote because it is already implemented and is ready for using.
4. PIO Remote is safe and works via SSL in Pro plan.

Thanks a lot that you use PlatformIO. We would be thankful if you support us with https://pioplus.com

P.S: Do you use PIO for commercial projects?

yveshanoulle, 2016-12-05:
2) As far as I understand PIO remote, the devices have to be connected to the internet. so yes that is forcing me when I want to work with your program to work in a way I don't want to.
3) As I said already multiple times, it's not possible as my arduino does not have enough memory as we speak.
4) I never said it was unsafe. I don't know what this is an answer to.
5) no, I want to use PIO to configure arduino's that will be lighting my house. Nothing commercial.
If it was commercial internet would not be a problem. These will be unsupported, left alone arduino's without internet connection.

ivankravets, 2016-12-05:
Where did you buy these boards?

yveshanoulle, 2016-12-05:
at a local shop.
Why?

ivankravets, 2016-12-05:
I mean are these boards from Arduino CC or ORG? I'll contact with vendor and ask to resovle this issue. Thanks.

yveshanoulle, 2016-12-05:
thanks http://www.arduino.org/ products/arduino-yun

ivankravets, 2016-12-06:
@artynet please take a look at this issue. You need to handle `upload_port`, if user specifies it to IP/mDNS , need to use OTA uploading. All things should be done in this file https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py

meelow, 2017-08-17:
I am also interested in this feature :-)

honsey, 2017-09-26:
I was searching for this as well, because I work with the LINKIT SMART DUO and want to upload via SSH.

I could come up with a first working proof of concept. Maybe its useful for some of you.. a bit hacky, but it works.

Add this env to platformio.ini
`[env:sshupload]
platform = atmelavr
board = yun
framework = arduino
extra_scripts = sshupload.py
upload_port = mylinkit.local`

sshupload.py (sitting next to platformio.ini in your project folder)
`Import("env")
import os

address = env.Dump("UPLOAD_PORT").replace('\'', '')

def ssh_upload(target, source, env):
    print "Attempting to upload firmware via SSH to %s" % address
    firmware = str(target[0])
    os.system('scp '+firmware+' root@'+address+':/root/firmware.hex')
    os.system('ssh root@'+address+' run-avrdude /root/firmware.hex')
    exit()
    #return False

env.AddPostAction("$BUILD_DIR/firmware.hex", ssh_upload)`

Trigger the SSH upload: `platformio run --target upload --environment sshupload` (You'll need to enter your YUN/LINKIT root password twice, if you didn't shared your SSH keys)

Let me know if it works on ARDUINO YUN as well. Don't have one, but should be similar.

mrcamilletti, 2018-02-02:
@hansipete it worked for me but i had to change target[0] with source[0].

hansipete, 2018-02-06:
@mrcamilletti cool – good to hear! i remember having also problems with target vs source[0]...

golyalpha, 2018-06-17:
Makes me feel like this should be a part of PlatformIO by now.

Meeh1, 2019-02-08:
Great, thank you! Worked for me too. 
Is there any way to store the root password so I don't have to enter it each time?

mrcamilletti, 2019-02-08:
> Great, thank you! Worked for me too.
> Is there any way to store the root password so I don't have to enter it each time?

You can install your ssh key with an 'ssh-copy-id' command and you won't need to type your password. [https://www.ssh.com/ssh/copy-id](url)