platformio/platform-atmelavr #24: Trinket 5V 8MHz timing issue

joe31093, 2016-12-10:
The timing on the Trinket seems to be off after downloading the code to the Trinket.  The Arduino build seems to be fine.  Here is the build output from the Arduino.  Please reference https://community.platformio.org/t/trinket-5v-8mhz-timing-issues/1149/6.

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/joeli/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/joeli/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/joeli/Documents/Arduino/libraries -fqbn=adafruit:avr:trinket3 -ide-version=10613 -build-path /var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/joeli/Documents/Arduino/Redirect/Redirect.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/joeli/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/joeli/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/joeli/Documents/Arduino/libraries -fqbn=adafruit:avr:trinket3 -ide-version=10613 -build-path /var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/joeli/Documents/Arduino/Redirect/Redirect.ino
Using board 'trinket3' from platform in folder: /Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10613 -DARDUINO_AVR_TRINKET3 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants/tiny8" "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10613 -DARDUINO_AVR_TRINKET3 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants/tiny8" "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp" -o "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10613 -DARDUINO_AVR_TRINKET3 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants/tiny8" "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp" -o "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp.o"
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.h:26:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:29,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:232,
                 from /var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Print.h:32:0: warning: "BIN" redefined
 #define BIN 2
 ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/iotn85.h:38:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:428,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:90,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:28,
                 from /var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/sketch/Redirect.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
 #define BIN     7

 ^
Compiling libraries...
Compiling core...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10613 -DARDUINO_AVR_TRINKET3 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants/tiny8" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/core/wiring_pulse.S.o"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10613 -DARDUINO_AVR_TRINKET3 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/joeli/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants/tiny8" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/core/WInterrupts.c.o"
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c: In function 'attachInterrupt':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:166:8: warning: #warning attachInterrupt may need some more work for this cpu (case 1) [-Wcpp]
       #warning attachInterrupt may need some more work for this cpu (case 1)
        ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c: In function 'detachInterrupt':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:255:8: warning: #warning detachInterrupt may need some more work for this cpu (case 1) [-Wcpp]
       #warning detachInterrupt may need some more work for this cpu (case 1)
        ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:29:0:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c: In function 'INT1_vect':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:310:15: warning: 'INT1_vect' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]
 IMPLEMENT_ISR(INT1_vect, EXTERNAL_INT_1)
               ^
[... cut ...] o.elf" "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/Redirect.ino.eep"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/Redirect.ino.elf" "/var/folders/dy/36kvbtrx123bh0zrzjrwc4sc0000gn/T/arduino_build_119630/Redirect.ino.hex"

Sketch uses 1,292 bytes (24%) of program storage space. Maximum is 5,310 bytes.
Global variables use 21 bytes of dynamic memory.

ivankravets, 2016-12-10:
1. You need to use `board = trinket3`
2. Try to replace variants in `~/.platformio/packages/framework-arduinoavr/variants` with [this](https://adafruit.github.io/arduino-board-index/boards/adafruit-avr-1.4.9.tar.bz2).

Does it work?

joe31093, 2016-12-11:
Changed the board type to trinket3 - no change in timing.
I replaced the variants with the link you provided and it did not make any change.

I will keep digging and see if I can see something.

ivankravets, 2016-12-11:
You can run PlatformIO in verbose mode via `pio run -v`. Try to compare both outputs.

ivankravets, 2017-01-27:
Could you try to update to the latest version? Via `pio update`. Could you reproduce this issue?

joe31093, 2017-01-29:
Updated Platform with latest Arduino Core.  Issue resolved.

closed 2017-01-29 by joe31093