platformio/platform-atmelavr #27: Lock To Specific Version

iwoloschin, 2017-01-23:
Is there any way to lock a platformio project to a specific atmelavr release?  For instance, I have a legacy project that gets updated about once a year, and it's historically had issues moving to new toolchain releases.  I'd like to get it working with the current release (well, right now it's working well with 1.3.1) and then lock my environment down so I can stop wasting time chasing toolchain issues when I need to push a release.

I've done something like this in my platformio.ini file:

`[env:uno]
platform: atmelavr@1.3.1
framework: arduino
board: uno`

This works great on a machine that's set up when 1.3.1 was latest, but it's failing now on a second machine now that there's a 1.4.1 release.  Is there any way to force an older atmelavr version to install on the second computer?

ivankravets, 2017-01-23:
Please sorry, fixed. Try again.

closed 2017-01-23 by ivankravets

iwoloschin, 2017-01-23:
Appears to be working now, thanks!  Much easier to separate code updates from toolchain updates now!