platformio/platform-atmelavr #28: Add Support for EnviroDIY Mayfly

SRGDamia1, 2017-01-24:
Add support for the EnviroDIY Mayfly:  https://github.com/EnviroDIY/Arduino_boards

{
  "build": {
    "extra_flags": "-DARDUINO_AVR_ENVIRODIY_MAYFLY",
    "core": "arduino",
    "f_cpu": "8000000L",
    "mcu": "atmega1284p",
    "variant": "mayfly"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "EnviroDIY Mayfly",
  "upload": {
    "maximum_ram_size": 16384,
    "maximum_size": 130048,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 57600
  },
  "url": "http://envirodiy.org/forums/",
  "vendor": "Stroud Water Research Center - EnviroDIY"
}

ivankravets, 2017-01-27:
Added https://github.com/platformio/platform-atmelavr/commit/7da2dc05198a6089f7ddb153aa70d1bf16cf0466

closed 2017-01-27 by ivankravets

ivankravets, 2017-01-27:
Does it work now without custom board?

Don't forget to run `pio update` before`

SRGDamia1, 2017-01-27:
It worked wonderfully!  Thank you!