platformio/platform-atmelavr #32: Bug: Upload on Arduino Uno fails

kalsan, 2017-02-19:
Moving issue https://github.com/platformio/platformio-core/issues/905 to here as requested.

### Configuration

**Operating system**: Arch Linux 64-bit

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.2.1 (installed in atom (with apm), not by the system's package manager packman or from AUR).

**Atom Version**: Tried 1.14.2-1 from official repos and from AUR.

### Description of problem
I cannot upload anything to Arduino Uno any more. avrdude will erase the old program but fails to upload the new one ("timeout"). I have tried several Uno boards. All of them work fine with Arduino IDE, but none works with PlatformIO since at least this week. I tried to upload a program that worked fine a few weeks ago and hasn't been changed since. An update broke the upload.

Serial Monitor is working.

#### Steps to Reproduce

1. Plug the board, open any PlatformIO Project
2. Press Upload

### Actual Results
Program on board gets erased, but the new program will not upload.

avrdude Output (ttyUSB0 is correct):
```
platformio run --target upload
[Sat Feb 18 10:15:57 2017] Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 28 compatible libraries
Looking for dependencies...
Project does not have dependencies

Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/megaatmega2560/firmware.hex
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude done.  Thank you.
 
 
*** [upload] Error 1
========================== [ERROR] Took 18.88 seconds ==========================
 
```

### If problems with PlatformIO Build System:

Not relevant, since the exact same config worked fine a few weeks ago.

ivankravets, 2017-02-20:
You are going to upload `megaatmega2560` firmware to Uno board. Please fix your `platformio.ini`.

closed 2017-02-20 by ivankravets

kalsan, 2017-02-20:
Oh noes. Can't believe I didn't figure that out in 2 hours of debugging...

thanks a lot and sorry for the unnecessary disturbance!