platformio/platform-atmelavr #33: Atmel AVR toolchain supports LTO only partially

zgoda, 2017-02-19:
Latest updates to AVR toolchain include LTO support in GCC, and compiler works indeed fine producing smaller object code. Unfortunately other tools such as `nm` (found in `$HOME/.platformio/packages/toolchain-atmelavr/bin/avr-nm`) does not support reading LTO object files, for example when trying to display symbols:

```
Print.o:
/home/jazg/.platformio/packages/toolchain-atmelavr/bin/avr-nm: Print.o: plugin needed to handle lto object
00000001 C __gnu_lto_slim
00000001 C __gnu_lto_v1
0000003e a __SP_H__
0000003d a __SP_L__
0000003f a __SREG__
00000000 a __tmp_reg__
00000001 a __zero_reg__
```

ladislas, 2017-11-26:
@zgoda If you want to use the latest toolchain from [osx-cross/avr](https://github.com/osx-cross/homebrew-avr), you can use this port of the platform: [osx-cross/platform-osxcrossavr](https://github.com/osx-cross/platform-osxcrossavr).

ivankravets, 2017-11-27:
Should be fixed in the latest toolchains.

closed 2017-11-27 by ivankravets