platformio/platform-atmelavr #34: Bug: Upload for Sparkfun Digital Sandbox fails

ybakos, 2017-02-20:
Via https://community.platformio.org/t/sparkfun-sandbox-programmer-not-responding/1499

## Configuration

Operating system: MacOS 10.11.6

PlatformIO Version (`platformio --version`): PlatformIO, version 3.2.1 (installed via the platformio  Atom download)

Atom Version: Tried 1.14.3 x64 from the platformio Atom download)

## Description of problem

Uploading a sketch using the Arduino IDE works fine, but not with platformio via atom nor the command line. avrdude fails to upload the program.

## Steps to Reproduce

Plug in the Sparkfun Digital Sandbox board, open the PlatformIO "blink" project.

platformio.ini is:

```
[env:sparkfun_digitalsandbox]
platform = atmelavr
board = sparkfun_digitalsandbox
framework = arduino
upload_port = /dev/cu.usbserial-A9048KU5
```

Press Upload

## Actual Results

Program compiles, but during the attempted upload, I see:

```
[Wed Feb 15 06:45:01 2017] Processing sparkfun_digitalsandbox (platform: atmelavr, upload_port: /dev/cu.usbserial-A9048KU5, board: sparkfun_digitalsandbox, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 28 compatible libraries
Looking for dependencies...
Project does not have dependencies
Linking .pioenvs/sparkfun_digitalsandbox/firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
928	      0	      9	    937	    3a9	.pioenvs/sparkfun_digitalsandbox/firmware.elf
Building .pioenvs/sparkfun_digitalsandbox/firmware.hex
Looking for upload port...
Use manually specified: /dev/cu.usbserial-A9048KU5
Uploading .pioenvs/sparkfun_digitalsandbox/firmware.hex
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
```

The attempts repeat until terminating after 10 failed attempts.

ivankravets, 2017-02-20:
Please test with
```
pio platform uninstall atmelavr
pio platform install https://github.com/platformio/platform-atmelavr/archive/develop.zip
```

closed 2017-02-20 by ivankravets

ybakos, 2017-02-21:
@ivankravets Thank you, the issue is indeed resolved. Enjoy the bounty!

ivankravets, 2017-02-21:
@ybakos Thanks! I've just released `atmelavr` development platform. Please run `pio update` command or in PIO IDE: Menu > PlatformIO > Update platforms, packages,....

commit 16f3ae3e (2017-02-20): Add new SparkFun boards; Update SparkFun board manifests // Resolve #34