platformio/platform-atmelavr #35: Arduino Leonardo COM port broken after an upload.

The00, 2017-02-21:
### Configuration

**Operating system**: Windows 7 64bit

**PlatformIO Version** (`platformio --version`): 3.2.1

### Description of problem
Hi !
When uploading code to an arduino leonardo, the normal serial port doesn't re-appears after uploading. Instead, the arduino stays on the bootloader serial port. 

#### Steps to Reproduce

1. create project for Eclipse IDE: `platformio init --ide eclipse --board leonardo`
2. Import project in Eclipse and add main.cpp source code
3. Upload a first time. It should work, but the serial port is lost. You will see instead  "Arduino leonardo bootloader" in the device manager.
4. Re-upload. It will fail.
5. You can unbrick the board by uploading a sketch using Arduino IDE(1.8.1). (set the port corresponding to Arduino leonardo bootloader). The normal serial port re-appears (Arduino Leonardo). 

### Actual Results
After an upload, the serial COM present is the bootloader, not the normal COM port. The uploaded code is running (blink demo code, tested with several delay). But the Serial communication doesn't work, and it's not possible to re-upload using Platformio.

### Expected Results
For a normal uploading, the IDE reset the board. The bootloader shows up, the code is uploaded, and then the normal serial port goes on. Here, it doesn't, even after manual reset.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#define WLED	13  // Most Arduino boards already have an LED attached to pin 13 on the board itself

void setup()
{
  pinMode(WLED, OUTPUT);  // set pin as output
  Serial.begin(9600);
}

void loop()
{
  digitalWrite(WLED, HIGH);
  delay(200);
  digitalWrite(WLED, LOW);
  delay(200);
  Serial.println("test");
}
```

### Additional info
 This problem looks like #178 , but according to the thread, the issue should have been solved in the 1.5.0 version. I tried with two Leonardo, same result on both.
I also tried on Ubuntu, and it's work, both re-upload and serial port. However, a board uploaded under Ubuntu still bug when connected to windows (same PC). To me it's seem related to a incompatibility issue with the window driver of the board.
What do you think ?

The00, 2017-02-24:
An other test I did : platformio + leonardo work fine on Ubuntu, but the leonardo flashed on Ubuntu bugs as previously back to W7.
Have you the same with yours OS ? ( flash on window 10, test with linux).

ivankravets, 2017-02-28:
Please re-test:

1. `pio platform uninstall atmelavr`
2. `pio platform install -U https://github.com/platformio/platform-atmelavr/archive/develop.zip`

If you still have a problem, please make first uploading with Arduino IDE.

closed 2017-02-28 by ivankravets

The00, 2017-02-28:
Thanks for your help !

I followed your step and I managed to get it work with some modification
The option -U seems to not exist, so I run it without.
The arduino core wasn't present in your package. I copied the cores directory from the arduino IDE into the package directory. Also removed the wiring_pulse.S file from it, it was gnerationg an error.

After that, everythings worked :)

Also, your package is quite different from the regular one, I guess it's used for developpement. 
Will the fix be added to the regular package ?

ivankravets, 2017-02-28:
Sorry, I wrote broken steps... I've just edited above. Please repeat.

andy-bren, 2019-11-12:
For the record, I have just experienced the same problem on a clean installation of VS Code + Platform IO v.4.10 (installed using the VS Code Extensions interface) with an Arduino Pro Micro (on Windows 10).

@ivankravets instructions above work great, however it took me a couple hours to track this issue down. Maybe some proactive work to prevent this for other users might be good.

commit 5583be63 (2017-02-28): Fix VID/PID for Arduino Leonardo // Resolve #35 Resolve #36