platformio/platform-atmelavr #36: Arduino Leonardo Uploading Problem

selcuksarii, 2017-02-23:
> OS : Manjaro Kde edition
> Board : Arduino Leonardo
> IDE : Atom with platformio

tried on both but same result..
> PlatformIO, version 3.3.0a11 --- Atom ide version
> PlatformIO, version 3.2.1 --- Linux CLI version

## Problem
I am trying to use atom-platformio-ide to program my arduino leonardo but there is a problem uploading the code;
the problem is, I can upload the code only once, after that for each trial platformio cant find the board. 
the only thing that lets the platformio to upload is upload some sample code from arduino ide, then platformio can upload for once again, but no more.

### i did the following to prepare the environment to run properly and there is no permission issue after this.
```
sudo chmod 666 /dev/ttyACM0
sudo usermod -a -G uucp
sudo usermod -a -G lock
```
## First Run
```
pio run --target upload --verbose
[Thu Feb 23 14:25:39 2017] Processing leonardo (platform: atmelavr, board: leonardo, framework: arduino)
----------------------------------------------------------------------------------------------------------
Collected 28 compatible libraries
Looking for dependencies...
Project does not have dependencies
BeforeUpload(["upload"], [".pioenvs/leonardo/firmware.hex"])
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
avrdude -v -p atmega32u4 -C /home/selcuk/.platformio/packages/tool-avrdude/avrdude.conf -c avr109 -b 57600 -P "/dev/ttyACM0" -D -U flash:w:.pioenvs/leonardo/firmware.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/home/selcuk/.platformio/packages/tool-avrdude/avrdude.conf"
User configuration file is "/home/selcuk/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : avr109
Overriding Baud Rate          : 57600
AVR Part                      : ATmega32U4
Chip Erase delay              : 9000 us
PAGEL                         : PD7                                                                                                                                                                  
BS2                           : PA0                                                                                                                                                                  
RESET disposition             : dedicated                                                                                                                                                            
RETRY pulse                   : SCK                                                                                                                                                                  
serial program mode           : yes                                                                                                                                                                  
parallel program mode         : yes                                                                                                                                                                  
Timeout                       : 200                                                                                                                                                                  
StabDelay                     : 100                                                                                                                                                                  
CmdexeDelay                   : 25                                                                                                                                                                   
SyncLoops                     : 32                                                                                                                                                                   
ByteDelay                     : 0                                                                                                                                                                    
PollIndex                     : 3                                                                                                                                                                    
PollValue                     : 0x53                                                                                                                                                                 
Memory Detail                 :                                                                                                                                                                      
                                                                                                                                                                                                     
Block Poll               Page                       Polled                                                                                                                                           
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack                                                                                                                    
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------                                                                                                                    
eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00                                                                                                                    
flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00                                                                                                                    
lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00                                                                                                                    
hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00                                                                                                                    
efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00                                                                                                                    
lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00                                                                                                                    
[... 60 lines cut ...]
avrdude -v -p atmega32u4 -C /home/selcuk/.platformio/packages/tool-avrdude/avrdude.conf -c avr109 -b 57600 -P "/dev/ttyACM0" -D -U flash:w:.pioenvs/leonardo/firmware.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/home/selcuk/.platformio/packages/tool-avrdude/avrdude.conf"
User configuration file is "/home/selcuk/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : avr109
Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "/dev/ttyACM0": Input/output error

avrdude done.  Thank you.

*** [upload] Error 1
===================================================================================== [ERROR] Took 7.74 seconds =====================================================================================
```

Any help would be appreciated

## Additional Notes :
>  Like arduino ide, Arduino Upload Package can upload properly.

The00, 2017-02-24:
I think I have the same issue (#35). Same behavior, except I have the bug on W7. I thought it may be OS dependent, but as you have it on a other OS..
Can you try on W7 ?

selcuksarii, 2017-02-24:
İ tried on W10 works as it should

The00, 2017-02-24:
An other test I did : platformio + leonardo work fine on Ubuntu, but the leonardo flashed on Ubuntu bugs as previously back to W7.
Have you the same with yours OS ? ( flash on window 10, test with linux).

selcuksarii, 2017-02-24:
I think my leonardo works fine if the upload is done without any error on both OS's. But i did not exactly understand what you mean by flash on Windows and test with linux. Do you mean serial communication ?

The00, 2017-02-24:
For me platformio doesn't upload exactly like the Arduino  toolchain, and it's seem to cause problem on certain OS. To see that, upload your code on the working OS with platformio. Then, switch to the other OS and connect the board and check if upload work with platformio.

selcuksarii, 2017-02-24:
My situation is not the same, only at Manjaro i cant upload twice, even if i upload on Windows, i can upload one more time at linux with platformio

ivankravets, 2017-02-28:
@selcuksarii @The00 

Please re-test:

1. `pio platform uninstall atmelavr`
2. `pio platform install -U https://github.com/platformio/platform-atmelavr/archive/develop.zip`

If you still have a problem, please make first uploading with Arduino IDE.

closed 2017-02-28 by ivankravets

p-hash, 2020-07-03:
@ivankravets Hello

I have the same issue both with default `atmelavr` and with yours.

```
> Executing task in folder 200704-001116-leonardo: C:\Users\Paul\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing leonardo (platform: atmelavr; board: leonardo; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/leonardo.html
PLATFORM: Atmel AVR 2.2.0 > Arduino Leonardo
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.0.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <sound>
Building in release mode
Checking size .pio\build\leonardo\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.5% (used 191 bytes from 2560 bytes)
Flash: [==        ]  24.7% (used 7076 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Auto-detected: COM9
Forcing reset using 1200bps open/close on port COM9
Waiting for the new upload port...
Uploading .pio\build\leonardo\firmware.hex
avrdude: ser_open(): can't open device "\\.\COM10": The system cannot find the file specified.

avrdude done.  Thank you.

*** [upload] Error 1
========================================================================================== [FAILED] Took 23.76 seconds ==========================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

selcuksarii, 2020-07-04:
@p-hash 
I think the problem resides in **\\.\COM10**, 
check the platformio.ini file for "upload_port" variable and set it to the port your board is connected to.
if you can't find that variable, add it to your boards settings, it should look similar to the one below.

`[env:uno]`
`platform = atmelavr`
`framework = arduino`
`upload_port = COM[10]`

P.S if you can post the content of the platformio.ini file, it will be easier to help.

p-hash, 2020-07-06:
@selcuksarii Hello
Here is my platformio.ini
```
[platformio]
default_envs = leonardo

[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
```
When I add `upload_port = COM[9]` to it, it fails with the same error as before, and when I add `upload_port = COM[10]`, it fails with following:
```
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
========================================================================================== [FAILED] Took 8.00 seconds ==========================================================================================
The terminal process terminated with exit code: 1
```

commit 5583be63 (2017-02-28): Fix VID/PID for Arduino Leonardo // Resolve #35 Resolve #36