platformio/platform-atmelavr #37: CurieTime library not found in PlatformIO Registry

matthewcarroll, 2017-02-24:
On PlatformIO 3.2.1 when building for an Arduino Yun, a warning is given saying the lib `CurieTime` cannot be found in the PlatformIO registry:

```
Processing yun (platform: atmelavr, lib_deps: CurieTime, board: yun, framework: arduino)
--------------------------------------------------------------------------------
Warning! Library `CurieTime` has not been found in PlatformIO Registry.
```

Importing the header and using the functions fails the build. How can the CurieTime lib be imported?

Thanks!

ivankravets, 2017-02-24:
This is a built-in library for https://github.com/platformio/platform-intel_arc32

If you want to use it for Atmel AVR, copy [this](https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieTime) library to "project/lib" folder. It should look like "project/lib/CurieTime/*files"

closed 2017-02-24 by ivankravets

matthewcarroll, 2017-02-24:
The Arduino web editor allows the import `#import <Time.h>` with a similar interface to the `CurieTime` lib (e.g. `hour()`, `minute()`). Importing it succeeds, but using the functions fails the build. How can the Time library be imported?

Thanks

ivankravets, 2017-02-24:
Ahhhh... Just use in `platformio.ini`
```
lib_deps = Time
```