platformio/platform-atmelavr #38: Modifying gcc-flags.json

matthewcarroll, 2017-02-28:
Modifying gcc-flags.json seems to have no effect on the build. Changing any setting in gcc-flags.json is unreflected in the verbose build output of a project (e.g. changing `gccDefaultCppFlags`). How can gcc-flags.json be enabled?

ivankravets, 2017-02-28:
http://docs.platformio.org/en/latest/projectconf.html#build-flags

closed 2017-02-28 by ivankravets

matthewcarroll, 2017-02-28:
Yes, I saw that. I thought that gcc-flags.json was editable given the instructions to edit gcc-flags.json for linting:
http://docs.platformio.org/en/latest/ide/atom.html?highlight=gcc-flags#smart-code-linter
http://docs.platformio.org/en/latest/ide/atom.html?highlight=gcc-flags#force-arduino-file-as-c

matthewcarroll, 2017-02-28:
Thanks. Yes, I saw that. I thought that gcc-flags.json was supposed to be editable given the instructions to edit it for linting:
http://docs.platformio.org/en/latest/ide/atom.html?highlight=gcc-flags#force-arduino-file-as-c
http://docs.platformio.org/en/latest/ide/atom.html?highlight=gcc-flags#smart-code-linter

ivankravets, 2017-02-28:
Thanks, I've just updated a docs:

* https://github.com/platformio/platformio-core/commit/ba750f0eee749863f827728c3725b8d5a14b0f63