platformio/platform-atmelavr #43: Add support for BOB3?

texjoachim, 2017-04-07:
Hi!

Would it be possible to add support for the BOB3 (http://www.bob3.org/)? It uses an Atmega88, I think and the necessary libraries are here: https://sourceforge.net/p/nibo/code/HEAD/tree/niborobolib/trunk/arduino16/

Regards,

Joe

ivankravets, 2017-04-23:
Please re-test with the latest development branch of dev platform (has not been released yet). Try dev version:

```
pio platform uninstall atmelavr
pio platform install https://github.com/platformio/platform-atmelavr.git
```

Does it work for you?

texjoachim, 2017-04-24:
Unfortunately not. I tried with the most simple example program:

`#include <BOB3.h>

 void setup() {
   bob3.setLed(EYE_1, WHITE);
 }`

Here is the error on build:
`

Bob3-Test
lib
src
BOB3-test.cpp
.gitignore
.travis.yml
platformio.ini
BOB3-test.cpp
 1
 1
 2
 3
 4
 5
 6
 7
 8
 9
10
11

/**¬
·*¬
·*¬
·*¬
·*/¬
·#include <BOB3.h>¬
¬
·void setup() {¬
···bob3.setLed(EYE_1, WHITE);¬
·}¬
 
1.3 s
platformio run
Compiling .pioenvs/bob3/lib/BOB3/bob3/analog.o
Compiling .pioenvs/bob3/lib/BOB3/bob3/ircom.o
Compiling .pioenvs/bob3/lib/BOB3/bob3/leds.o
Compiling .pioenvs/bob3/lib/BOB3/bob3/main.o
Archiving .pioenvs/bob3/libFrameworkArduino.a
Compiling .pioenvs/bob3/lib/BOB3/bob3/time.o
Indexing .pioenvs/bob3/libFrameworkArduino.a

/home/joachim/.platformio/packages/framework-arduinoavr/libraries/__cores__/nicai/BOB3/src/bob3/time.c: In function 'delay_tick
s':
/home/joachim/.platformio/packages/framework-arduinoavr/libraries/__cores__/nicai/BOB3/src/bob3/time.c:40:5: warning: implicit 
declaration of function 'yield' [-Wimplicit-function-declaration]
yield();
^
Archiving .pioenvs/bob3/lib/libBOB3.a
Indexing .pioenvs/bob3/lib/libBOB3.a
Linking .pioenvs/bob3/firmware.elf
/tmp/ccT0BUti.ltrans0.ltrans.o: In function `main':
ccT0BUti.ltrans0.o:(.text.startup+0x4e): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pioenvs/bob3/firmware.elf] Error 1
========================== [ERROR] Took 0.99 seconds ==========================
 
PIO Buildsrc/BOB3-test.cpp00010:3(5, 67)
LFUTF-8C++
`

valeros, 2017-05-02:
Hi @texjoachim !
Looks like you sketch doesn't contain the`loop` function. Here is a simple blink sketch for your board:
```cpp
#include <Arduino.h>
#include <BOB3.h>

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}
```
Does it work?

texjoachim, 2017-05-02:
Ok, this one compiles perfectly, thanks!

However, I get an error message when I try to upload the sketch:

    platformio run --target upload
    [Tue May  2 20:42:50 2017] Processing bob3 (platform: atmelavr, board: bob3, framework: arduino)
    --------------------------------------------------------------------------------
    Verbose mode can be enabled via `-v, --verbose` option
    Collected 31 compatible libraries
    Looking for dependencies...
    Library Dependency Graph
    |-- <BOB3> v1.0.0
    
    Looking for upload port...
    Uploading .pioenvs/bob3/firmware.hex
    avrdude: ser_open(): can't open device "com1": No such file or directory
    avrdude done.  Thank you.
     
     
    *** [upload] Error 1

Dmesg on my system says:

    [  613.508105] usb 3-3: new full-speed USB device number 5 using xhci_hcd
    [  613.637875] usb 3-3: New USB device found, idVendor=16c0, idProduct=0933
    [  613.637877] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [  613.637878] usb 3-3: Product: ProgBob BOB3-programmer
    [  613.637879] usb 3-3: Manufacturer: nicai-systems
    [  613.637881] usb 3-3: SerialNumber: 00000
    [  613.638254] cdc_acm 3-3:1.0: ttyACM0: USB ACM device

I've updated the platformio.ini to:

    [env:bob3]
    platform = atmelavr
    board = bob3
    framework = arduino
    upload_port = /dev/ttyACM0

But it still complains about COM1 which, being a Linux system, doesn't exist.

ivankravets, 2017-05-03:
Please 

1. `pio update`
2. Remove `upload_port` from your platformio.ini. PIO should detect port automatically.
3. Try uploading again.

texjoachim, 2017-05-03:
This is an improvement as it now finds the correct port.

    Looking for upload port...
    Auto-detected: /dev/ttyACM0
    *** [upload] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'

However, now it can't open it. Here are the details on the port:

    crw-rw---- 1 root dialout 166, 0  3. Mai 20:32 /dev/ttyACM0

Needless to say I'm in group dialout so the permissions should be alright.

ivankravets, 2017-05-03:
Could you try to reconnect this board?

texjoachim, 2017-05-03:
I did that.
No change. :(

ivankravets, 2017-05-03:
Cu

texjoachim, 2017-05-03:
No, it is only possible to program this board with some website.
That's one of the reasons I wanted to use Atom which I use for my UNOs, too.

ivankravets, 2017-05-03:
OK, does it work wit that site? Maybe, that site via Google Chrome's Serial uses your board

1. Reboot your machine
2. Don't open Chrome.
3. Try PIO

texjoachim, 2017-05-03:
The problem with that site (https://www.progbob.org/) is that it uses a special connector program (dude.bob3.org) which you have to compile yourself (which fails on openSUSE). It does not use Chrome (which I don't even have installed.)

ivankravets, 2017-05-03:
Could we close this issue?

texjoachim, 2017-05-03:
If there is no hope left to get it to work, of course.

ivankravets, 2017-05-03:
You should decide which software to use. That is not our requirements. If your serial device is occupied by another program, we can not use it anymore while that program release it.

If you would like to sue PIO, then don't run that "special connector program ".

texjoachim, 2017-05-03:
I don't run that program as I didn't even compile it. As far as I know, no program is using a serial device.

texjoachim, 2017-05-03:
So I would really like to use PIO.

ivankravets, 2017-05-03:
Please sorry, I've not got you. Let's forget about PIO.

Could you try this command in system terminal?
```
screen /dev/ttyACM0
```

Do you see any error?

texjoachim, 2017-05-04:
I get ...
    [screen is terminating]

and sometimes I get nothing, that is the cursor is away and the terminal window just stays blank.

commit 9c6c1075 (2017-04-23): New boards by Nicai Systems // Resolve #43
commit 4d787080 (2017-05-03): Require upload_port for Nicai boards // Resolve #43