platformio/platform-atmelavr #44: Arduino I2C Wire no data

Thijxx, 2017-04-13:
**Versions**
IDE: 17.3
Core: 3.2.1

**Config**
```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
lib_deps =
  Wire
```
I am trying to move from the Arduino IDE to platformio and that is not as simple as copy/paste but I am getting there. The issue I run into at the moment is an I2C device (LM75) using the Wire.h library that returns no data.  I have taken out as many bits and pieces as possible to isolate the issue.

```
#include <Arduino.h>
#include <Wire.h>

void setup()
{
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
}

void loop()
{
  Wire.requestFrom(0x48, 2);    // request 6 bytes from slave device #2

  while(Wire.available())    // slave may send less than requested
  {
    word c = Wire.read();    // receive a byte as character
    float t = ((float)(int)c / 32) / 8;
    Serial.println(String(t));         // print the character
  }

  delay(500);
}
```
This exact code works when I use the Arduino IDE but not when I use platformio. In the past this was due to difference in pin names. Here, since it is using I2C this will not be the case.

I have use an I2C Scanner and the code finds the chip and the address is as it should be, 0x48. But the request or read from wire gives an other result when I built with platformio than when I build with Arduinol IDE.

I have taken the Arduino Wire library and overwritten the platformio version (same code) with this just to be sure it's equal.

ivankravets, 2017-07-23:
Does it work with the latest updates?

Thijxx, 2017-07-24:
I can setup a test tonight, will update here.

Thijxx, 2017-07-24:
The LM75 works great in platformio now :) awesome.
Also tried it on the ATtiny85 again, still nothing, but thats another story.

Updated code, added line with `Wire.begin()`

```
#include <Arduino.h>
#include <Wire.h>
#include <LM75.h>

LM75 sensor;

void setup() {
  Wire.begin();
  Serial.begin(9600);
}

void loop() {
  // get temperature from sensor
  Serial.print("Current temp: ");
  Serial.print(sensor.temp());
  Serial.println(" C");

  // shutdown the sensor and wait a while
  sensor.shutdown(true);
  delay(3000);
  // wake up sensor for next time around
  sensor.shutdown(false);
  delay(1000);

  Serial.println();
}
```

closed 2017-07-24 by Thijxx