platformio/platform-atmelavr #46: ATTiny85 build or RadioHead problem?

Thijxx, 2017-04-17:
The RadioHead library does not work with ATTiny85 with protocol stk500v1.
This works from Arduino IDE.

These settings do not build. They fail over `RadioHead_ID124/RH_RF24.cpp:413:3: error: 'Serial' was not declared in this scope`
```
[env:myenv]
platform = atmelavr
framework = arduino
board = attiny85
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port = /dev/ttyUSB0
upload_speed = 19200
lib_deps = RadioHead
lib_ignore = SPI
```

With these settings it builds, is also an ATTiny85 but an other board with built-in USB and more.
```
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
lib_deps = RadioHead
lib_ignore = SPI
```
While these settings for the digispark work, uploading to my ATTiny85 is not possible because I use stk500v1 and the digispark some USB connection. So I cannot trick it into building and uploading to my ATTiny85 using digispark settings.

```
#include <RH_ASK.h>

#define R_PIN PB4
#define T_PIN PB1

RH_ASK driver(2000, R_PIN, T_PIN);

void setup() {
}

void loop()
{
  const char *msg = "hello";

  driver.send((uint8_t *)msg, strlen(msg));
  driver.waitPacketSent();
  delay(200);
}
```

ivankravets, 2017-04-23:
Could you try?
```
[env:myenv]
platform = atmelavr
framework = arduino
board = attiny85
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port = /dev/ttyUSB0
upload_speed = 19200
lib_deps = RadioHead
lib_ignore = SPI
build_flags = -DARDUINO_ARCH_AVR
```

Thijxx, 2017-04-24:
Thanks for your reply, I've tried it but it did not help. The error stays the same.

I had so many issues with platformio that I moved back to the Arduino IDE so I close this issue since I cannot test it anymore.

closed 2017-04-24 by Thijxx

ivankravets, 2017-04-24:
* Arduino IDE > Settings > Verbose mode for Build/Upload
* `pio run --verbose`

You can compare 2 build processes (flags). Please share them on http://pastebin.com.

I don't see a problem why it can't work with PIO too.