platformio/platform-atmelavr #47: ATTiny Support

chappy1978, 2017-04-25:
ts seems there are a few bugs when programming ATTiny's using Platformio and the standard atmelavr packages. (Example: warning: "BIN" redefined #define BIN 2) I know its just a warning but who wants to look at that every time they build.  I thought it would be nice to use ATTinyCore. After looking through Platformio's folders the folder format is very different, as compared to Arduino. How would I go through creating a package install for Platformio?  

https://github.com/SpenceKonde/ATTinyCore

ivankravets, 2017-04-28:
@valeros Please push all changes to https://github.com/platformio/platformio-pkg-framework-arduinoavr

* Add new core named `attiny`
* Updated existing board manifest with `"core": "attiny"`
* Add new variants and remove obsolete (currently added for existing attiny)
* Merge https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/avrdude.conf with our config in our packages for all architectures
* Place https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/libraries to `__cores__/attiny`

ivankravets, 2017-05-03:
@chappy1978 @Thijxx could you re-test it?

1. Uninstall existing stable version of Atmel AVR development platform via `pio platform uninstall atmelavr`
2. Install development version: `pio platform install https://github.com/platformio/platform-atmelavr.git`

See all board manifests: https://github.com/platformio/platform-atmelavr/tree/develop/boards

Does it work?

Thijxx, 2017-05-03:
I can see the changes made but I have no clue how to select the board.

```
[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
```

This is not it, maybe you could shine a light on the config or better, point me to the right spot in the documentation :D

ivankravets, 2017-05-03:
@Thijxx Could you provide a demo project to reproduce this issue?

Thijxx, 2017-05-03:
This is the project folder.

[attiny85_spencekondetest.zip](https://github.com/platformio/platform-atmelavr/files/974715/attiny85_spencekondetest.zip)

valeros, 2017-05-04:
Hi @Thijxx! Probably you should use software SPI as mentioned in the RadionHead library:
```
lib/TinyHead/RadioHead.h:786:4: warning: #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
```
In ATTiny Core [README](https://github.com/SpenceKonde/ATTinyCore#spi-support) said that with `ATtiny x5 (25/45/85)` you can use SPI USI library.

Have you tried to build your project in Arduino IDE?

Thijxx, 2017-05-04:
Yes, this works in the Arduino IDE.
Below is the verbose compile log fromt the Arduino IDE based on the same code. There are many warning but no errors.

```
/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/thijs/.arduino15/packages -hardware /home/thijs/Documents/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/thijs/.arduino15/packages -libraries /home/thijs/Documents/Arduino/libraries -fqbn=ATTinyCore-master:avr:attinyx5:LTO=disable,TimerClockSource=default,chip=85,clock=8internal,bod=disable -ide-version=10802 -build-path /tmp/arduino_build_937199 -warnings=default -build-cache /tmp/arduino_cache_229451 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_875097/Blink.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/thijs/.arduino15/packages -hardware /home/thijs/Documents/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/thijs/.arduino15/packages -libraries /home/thijs/Documents/Arduino/libraries -fqbn=ATTinyCore-master:avr:attinyx5:LTO=disable,TimerClockSource=default,chip=85,clock=8internal,bod=disable -ide-version=10802 -build-path /tmp/arduino_build_937199 -warnings=default -build-cache /tmp/arduino_cache_229451 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_875097/Blink.ino
Using board 'attinyx5' from platform in folder: /home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr
Using core 'tiny' from platform in folder: /home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr
Detecting libraries used...
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "/tmp/arduino_build_937199/sketch/Blink.ino.cpp" -o "/dev/null"
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/TinyHead" "/tmp/arduino_build_937199/sketch/Blink.ino.cpp" -o "/dev/null"
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/TinyHead" "/home/thijs/Documents/Arduino/libraries/TinyHead/RHCRC.cpp" -o "/dev/null"
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/TinyHead" "/home/thijs/Documents/Arduino/libraries/TinyHead/RHGenericDriver.cpp" -o "/dev/null"
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/TinyHead" "/home/thijs/Documents/Arduino/libraries/TinyHead/RH_ASK.cpp" -o "/dev/null"
Generating function prototypes...
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/TinyHead" "/tmp/arduino_build_937199/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_937199/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin/arduino-ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_937199/preproc/ctags_target_for_gcc_minus_e.cpp"
[... cut ...]
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -R .eeprom  "/tmp/arduino_build_937199/Blink.ino.elf" "/tmp/arduino_build_937199/Blink.ino.hex"
Using library TinyHead in folder: /home/thijs/Documents/Arduino/libraries/TinyHead (legacy)
Sketch uses 5048 bytes (61%) of program storage space. Maximum is 8192 bytes.
Global variables use 393 bytes (76%) of dynamic memory, leaving 119 bytes for local variables. Maximum is 512 bytes.
Low memory available, stability problems may occur.
```

valeros, 2017-05-04:
Could you please try to manually disable SPI library in your project?
```ini
[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
lib_ignore = SPI
```

Thijxx, 2017-05-04:
This seems to work. Could you please elaborate on why this works the way it does? I'd expect the Arduino IDE to show the same behaviour..

```
[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
lib_ignore = SPI
```

How the core switching works is still not clear to me, don't I need to include `core = tiny` or anything in the `ini` file?

valeros, 2017-05-04:
Actually, there is no `core` option in `platformio.ini` at all. `core` option is specified in the board configuration file `attiny85.json` and you cannot change it in your `platformio.ini`.
As for the error with `SPI` library, there is no default `SPI` implementation for `tiny` core, so PlatformIO build system decided to compile a generic one for all `AVR` boards. We will fix this behavior in the next framework update.

Thijxx, 2017-05-04:
@ivankravets, you may want to close this issue again. Thanks.

ivankravets, 2017-05-04:
I've just deployed new Arduino framework. @Thijxx could you try `pio update`, clean project via `pio run -t clean` and build again with commented this line `lib_ignore = SPI`.

Does it work now?

Thijxx, 2017-05-04:
If I comment the lib_ignore so it looks like `;lib_ignore = SPI` I get the same error again.

ivankravets, 2017-05-04:
@Thijxx could you try keeping this line commented and add 
```
lib_ldf_mode = chain+
```

See http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

Thijxx, 2017-05-04:
This works, so does it mean I need to doublecheck the library for an IF statement that still points to the SPI library?

ivankravets, 2017-05-05:
Now, please remove both lines `lib_ldf_mode` and `lib_ignore`. Run `pio update`, clean project and rebuild.

Thijxx, 2017-05-06:
🎉 awesome, thanks y'all

Thijxx, 2018-03-12:
This is back again, please re-open ticket and re-repair.
The solution is adding `lib_ldf_mode = chain+` again.

glucee, 2018-08-08:
I have tried the Spark Attiny AVR Programmer, it works well. You could check my [repo](https://github.com/glucee/attiny85_platformio) to try my config.

hwacookie, 2018-11-27:
Until this is fixed in platformio (which I really love, and I really do appreciate the work @ivankravets and his fellows are putting into it), just a hint for anyone stumbling upon this like I did: I can confirm that compiling the exact same source-code *does* work in the Arduino IDE.

eddnshoulders, 2019-11-18:
Just in case this has been forgotten or not marked as closed on the last fix, I've just had this same problem with arduino framework 4.1.2 on core 4.1.0.

Thanks to those above for providing a workaround.