platformio/platform-atmelavr #49: avrdude hangs frequently while programming leonardo devices

Rantanen, 2017-05-03:
When running `pio run -t upload` avrdude hangs often on lower performance hosts.

The leonardo-styled target devices use `wait_for_upload_port = true` for programming. On Linux (at least the Raspbian) the device file stays the same before and after the reset - thus the wait-for-new-port functionality will end up waiting all the way up to the 5-second timeout. This matches the functionality of the Arduino IDE.

However before the 5-second timeout, PlatformIO waits an extra second before starting to pull the serial ports. This takes PlatformIO's total wait during the wait_for_upload_port to 6 seconds.

In addition to that there is roughly half a second of other delays before avrdude is launched.

This leaves only 1.5 seconds for avrdude to start the programming sequence before the 8 seconds the bootloader waits expire and the bootloader starts running the program on the board.

This 1.5 seconds isn't enough time for slower host machines, such as the Raspberry Pi Zero.

ivankravets, 2017-05-03:
I've just removed this extra second from PIO Core: https://github.com/platformio/platformio-core/commit/30ff491a34b47fd0adc4c7395fa86b80b7919e64

Could you try dev build?

```
pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
```

Rantanen, 2017-05-03:
Works way better. Removing the 1 second delay made the upload around 5.5 seconds faster :)

After the reset it seems that RPiZero takes around 500 ms to re-initialize the port. This caused following events to happen during the loop.

```
0 ms:    PlatformIO: Store current ports, reset and sleep for 1 second.
100 ms:  Raspberry: /dev/ttyAMC0 disappears due to reset
400 ms:  Raspberry: /dev/ttyAMC0 reappears after the reset
1000 ms: PlatformIO: Check the ports. Upload thinks the /dev/ttyAMC0 it sees is the old one.
6000 ms: Timeout on the loop, uses /dev/ttyAMC0
```

Removing the 1 sec delay allows PlatformIO to recognize that the port disappeared after the reset and then around 500 ms it correctly recognizes the new port and breaks out of the loop:

```
0 ms:    PlatformIO: Store current ports, reset and sleep for 1 second.
100 ms:  Raspberry: /dev/ttyAMC0 disappears due to reset
250 ms:  PlatformIO checks the ports. Recognizes /dev/ttyAMC0 is gone.
400 ms:  Raspberry: /dev/ttyAMC0 reappears after the reset
500 ms:  PlatformIO checks the ports. Sees the new /dev/ttyAMC0 and uses that.
```

Definitely keep the fix. :)

closed 2017-05-03 by Rantanen

ivankravets, 2017-05-03:
@Rantanen thanks a lot for detailed info and report!