platformio/platform-atmelavr #53: Debug probe support for atmel-ice / dragon and atmega328p processor

warrenwoolseyiii, 2017-06-19:
As the title states I am looking for debug probe support for the atmel-ice / and or dragon debug probes. I specifically need support for the atmega328p processor which I am using in a custom application.

valeros, 2017-06-20:
Hi @warrenwoolseyiii ! Thanks for your request!
We don't have required hardware, so it would be great if you could help us implement initial support for debugging in `atmelavr` platform.
First steps:
Install the development version of `atmelavr`:
```
pio platform uninstall atmelavr
pio platform install https://github.com/platformio/platform-atmelavr.git
```
Then create a new project and modify `platformio.ini` to fit your needs, e.g.:
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
debug_port = :4242
debug_tool = custom
debug_init_cmds =
  target remote $DEBUG_PORT
  file "$PROG_PATH"
  load
  tb main
debug_server =
  /path/to/avarice
  --edbg
  --debugwire 
  --ignore-intr 
  :4242
```
Now you can start debugging session from terminal using `pio debug --interface gdb`
Some of above commands might now work, especially `load` (in this case you will need to upload debug version of firmware manually using `avrdude`)

warrenwoolseyiii, 2017-06-20:
valerii this is awesome thank you so much for getting me started. I will
keep you updated with progress.

albertskog, 2018-01-18:
How's the progress @warrenwoolseyiii, did you manage to get it working? 😄

kasbah, 2018-03-03:
I tried this and got:

```
$ pio debug --interface gdb -x .pioinit
PlatformIO Plus (https://pioplus.com) v0.14.1
Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/kaspar/.local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "cmd_debug.py", line 61, in cmd_debug.cli
  File "cmd_debug.py", line 66, in cmd_debug.cli
  File "cmd_debug.py", line 235, in cmd_debug.validate_debug_options
TypeError: argument of type 'NoneType' is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================
```

```
$ pio --version
PlatformIO, version 3.5.2rc2
$ pio update
...
Platform Atmel AVR
--------
Updating atmelavr                        @ 5d08019        [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating framework-arduinoavr            @ 1.10620.2      [Up-to-date]
...
```

With PlatformIO 3.5.1 I get:

```
$ pio debug --interface gdb -x .pioinit
************************************************************************************
Obsolete PIO Core v3.5.1 is used (previous was 3.5.2rc2)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
************************************************************************************
PlatformIO Plus (https://pioplus.com) v0.14.1
Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/tmp/venv/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/tmp/venv/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/tmp/venv/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/tmp/venv/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/tmp/venv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/tmp/venv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "cmd_debug.py", line 61, in cmd_debug.cli
  File "cmd_debug.py", line 66, in cmd_debug.cli
  File "cmd_debug.py", line 233, in cmd_debug.validate_debug_options
AttributeError: 'PlatformBoardConfig' object has no attribute 'get_debug_tool_name'

...
```

suroh001, 2019-06-25:
Any updates on this?

lloydjatkinson, 2020-01-03:
Considering its now 2020 and the popularity of the 328/Arduino, I'm quite surprised debugging isn't supported yet. Is this being considered or not?

audryhome, 2020-03-07:
Since Black Magic probe is supported, I think it may be possible to hack this firmware to make it pio friendly with AVR ... any other idea?

fatihelbl, 2022-02-24:
Hello, I have PICKIT4. I want debugging . Arduino leonardo is also trying to make mistakes. But it didn't work out. How should I proceed? 
I have Win10, Visual Studio Code( 1.64.2), PlatformIO (v2.4.2)