platformio/platform-atmelavr #55: Wrong upload COM port detected

AndreKR, 2017-06-23:
PlatformIO 0.3.0 for VSCode

I'm trying to upload to a Sparkfun Pro Micro.

The Pro Micro upload process is the same as with the Arduino Leonardo: Open the selected COM port with 1200 baud and immediately close it to trigger a reset. Then wait until a new COM port appears (can be the same or a different number), then run a normal Arduino upload over this COM port.

In Arduino IDE it works, in PlatformIO for some reason it tries to upload to COM13 even though the new COM port is COM25:

**Log:**
```
[06/23/17 22:04:54] Processing sparkfun_promicro16 (platform: atmelavr; upload_port: COM16; board: sparkfun_promicro16; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Collected 27 compatible libraries
Looking for dependencies...
Project does not have dependencies
Looking for upload port...
Use manually specified: COM16
Forcing reset using 1200bps open/close on port COM16
Waiting for the new upload port...
Uploading .pioenvs\sparkfun_promicro16\firmware.hex
avrdude: ser_open(): can't open device "\\.\COM13": Zugriff verweigert
avrdude done.  Thank you.
*** [upload] Error 1
 [ERROR] Took 4.18 seconds 
```

**Pre-reset:**
[image]

**Post-reset:**
[image]

**platformio.ini:**
```
[env:sparkfun_promicro16]
platform = atmelavr
board = sparkfun_promicro16
framework = arduino
upload_port = COM16
```
(Curiously if I don't specify any upload_port the reset is still triggered.)

ivankravets, 2017-06-23:
Could you provide an output of `pio device list` command?

AndreKR, 2017-06-23:
**In normal mode:**
```
COM13
-----
Hardware ID: USB VID:PID=0BDB:1911 LOCATION=1-1.4.4
Description: F5521gw Mobile Broadband Device Management (COM13)

COM6
----
Hardware ID: PCI\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\3&21436425&0&B3
Description: Intel(R) Active Management Technology - SOL (COM6)

COM14
-----
Hardware ID: ERICSSON\GPS\B113C87391F2FED0_09
Description: F5521gw Mobile Broadband GPS Port (COM14)

COM15
-----
Hardware ID: BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\8&2D4DBFD6&0&000000000000_00000004
Description: Standardmäßgige Seriell-über-Bluetooth-Verbindung (COM15)

COM16
-----
Hardware ID: USB VID:PID=2341:8037 LOCATION=1-1.5.5.5
Description: Arduino Micro (COM16)
```

**In bootloader:**
```
COM13
-----
Hardware ID: USB VID:PID=0BDB:1911 LOCATION=1-1.4.4
Description: F5521gw Mobile Broadband Device Management (COM13)

COM6
----
Hardware ID: PCI\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\3&21436425&0&B3
Description: Intel(R) Active Management Technology - SOL (COM6)

COM14
-----
Hardware ID: ERICSSON\GPS\B113C87391F2FED0_09
Description: F5521gw Mobile Broadband GPS Port (COM14)

COM15
-----
Hardware ID: BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\8&2D4DBFD6&0&000000000000_00000004
Description: Standardmäßgige Seriell-über-Bluetooth-Verbindung (COM15)

COM24
-----
Hardware ID: USB VID:PID=2341:0037 SER=7 LOCATION=1-1.5.5
Description: Arduino Micro bootloader (COM24)
```

ivankravets, 2017-06-23:
Please create `boards` folder in a project root folder and place here `sparkfun_promicro16.json` file with the next contents:
```json
{
  "build": {
    "core": "arduino",
    "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_PROMICRO16",
    "f_cpu": "16000000L",
    "hwids": [
      [
        "0x2341",
        "0x8037"
      ],
      [
        "0x2341",
        "0x0037"
      ]
    ],
    "mcu": "atmega32u4",
    "usb_product": "SparkFun Pro Micro",
    "variant": "sparkfun_promicro"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "SparkFun Pro Micro 5V/16MHz",
  "upload": {
    "disable_flushing": true,
    "maximum_ram_size": 2560,
    "maximum_size": 28672,
    "protocol": "avr109",
    "require_upload_port": true,
    "speed": 57600,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true
  },
  "url": "https://www.sparkfun.com/products/12640",
  "vendor": "SparkFun"
}
```

Does it work now?

ivankravets, 2017-06-23:
Also, please remove `upload_port = COM16` from platformio.ini config

AndreKR, 2017-06-23:
Hm, this seems to be a clone, because on https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide#faq-vid it says:

> All SparkFun ATmega32U4 boards share the same VID – 0x1B4F, and they all have unique PIDs. 5V Pro Micros lay claim to PIDs 0x9205 and 0x9206 (one for the bootloader, one for the sketch). 3.3V Pro Micros will show up as 0x9203 and 0x9204 for bootloader and sketch, respectively. And the Fio v3 has 0xF100 and 0xF101.

In the Arduino IDE this pair of VID/PID is recognized as "Arduino/Genuino Micro".

AndreKR, 2017-06-23:
Ok, so with your file the upload works.

Out of curiosity I tried with "Arduino Micro" (ID: `micro`) as board, but that fails, what's the VID/PID for that?

Yeah, I don't know... add "eBay Pro Micro" to the board database? :)

AndreKR, 2017-06-23:
Hm, wait a sec... `%USERPROFILE%/.platformio/platforms/atmelavr/boards/micro.json` does actually contain
```
"hwids": [
      [
        "0x2341", 
        "0x0037"
      ], 
      [
        "0x2341", 
        "0x8037"
      ], 
      [
        "0x2A03", 
        "0x0037"
      ], 
      [
        "0x2A03", 
        "0x8037"
      ]
    ], 
```

Why didn't it work with the board set to `micro`.

AndreKR, 2017-06-23:
Oh, sorry

> Ok, so with your file the upload works.

That was WRONG, the upload still doesn't work. Same error.

ivankravets, 2017-06-23:
Please open PlatformIO IDE Terminal in VSCode (see icon in toolbar, bottom) and run:

```
pip install -U https://github.com/platformio/platformio-core/archive/feature/debug-avr-micro.zip
```

Does it work now?

AndreKR, 2017-06-23:
Nope, same thing (COM13).

AndreKR, 2017-06-23:
I think the problem might have nothing to do with the board but rather with my WAN module (F5521gw) or the way the USB devices are queried.

I found `pioupload.py` in `%USERPROFILE%\.platformio\penv\Lib\site-packages\platformio\builder\tools`.

There I see that the VID/PID is not used at all for choosing the upload COM port but rather it waits for a new COM port, just as the Arduino IDE does. Unlike the Arduino IDE (I think) it compares the COM ports not only by number, but also by a field named `hwid`.

I added `pprint(before)`, `pprint(now)` and `pprint(new_port)`, and I found:
```
'before:'
[{'description': u'F5521gw Mobile Broadband Device Management (COM13)',
'hwid': 'USB VID:PID=0BDB:1911 LOCATION=1-1.4.4',
'port': u'COM13'},
{'description': u'Intel(R) Active Management Technology - SOL (COM6)',
'hwid': u'PCI\\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\\3&21436425&0&B3',
'port': u'COM6'},
{'description': u'F5521gw Mobile Broadband GPS Port (COM14)',
'hwid': u'ERICSSON\\GPS\\B113C87391F2FED0_09',
'port': u'COM14'},
{'description': u'Standardm\xe4\xdfgige Seriell-\xfcber-Bluetooth-Verbindung (COM15)',
'hwid': u'BTHENUM\\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\\8&2D4DBFD6&0&000000000000_00000004',
'port': u'COM15'},
{'description': u'Arduino Micro (COM26)',
'hwid': 'USB VID:PID=2341:8037 LOCATION=1-1.5.5.5',
'port': u'COM26'}]

'now:'
[{'description': u'F5521gw Mobile Broadband Device Management (COM13)',
'hwid': 'USB VID:PID=0BDB:1911 LOCATION=1-1.4.4',
'port': u'COM13'},
{'description': u'Intel(R) Active Management Technology - SOL (COM6)',
'hwid': u'PCI\\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\\3&21436425&0&B3',
'port': u'COM6'},
{'description': u'F5521gw Mobile Broadband GPS Port (COM14)',
'hwid': u'ERICSSON\\GPS\\B113C87391F2FED0_09',
'port': u'COM14'},
{'description': u'Standardm\xe4\xdfgige Seriell-\xfcber-Bluetooth-Verbindung (COM15)',
'hwid': u'BTHENUM\\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\\8&2D4DBFD6&0&000000000000_00000004',
'port': u'COM15'},
{'description': u'Arduino Micro (COM26)',
'hwid': 'USB VID:PID=2341:8037 LOCATION=1-1.5.5.5',
'port': u'COM26'}]

'now:'
[{'description': u'F5521gw Mobile Broadband Device Management (COM13)',
'hwid': 'USB VID:PID=0BDB:1911',
'port': u'COM13'},
{'description': u'Intel(R) Active Management Technology - SOL (COM6)',
'hwid': u'PCI\\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\\3&21436425&0&B3',
'port': u'COM6'},
{'description': u'F5521gw Mobile Broadband GPS Port (COM14)',
'hwid': u'ERICSSON\\GPS\\B113C87391F2FED0_09',
'port': u'COM14'},
{'description': u'Standardm\xe4\xdfgige Seriell-\xfcber-Bluetooth-Verbindung (COM15)',
'hwid': u'BTHENUM\\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\\8&2D4DBFD6&0&000000000000_00000004',
'port': u'COM15'},
{'description': u'Arduino Micro (COM26)',
'hwid': 'USB VID:PID=2341:8037',
'port': u'COM26'}]

'new_port:'
u'COM13'
```

As you can see, the WAN module is reported two times as
```
{'description': u'F5521gw Mobile Broadband Device Management (COM13)',
'hwid': 'USB VID:PID=0BDB:1911 LOCATION=1-1.4.4',
'port': u'COM13'}
```
but then as:
```
{'description': u'F5521gw Mobile Broadband Device Management (COM13)',
'hwid': 'USB VID:PID=0BDB:1911',
'port': u'COM13'}
```

So `p not in before` becomes true and it erroneously selects COM13.

ivankravets, 2017-06-23:
Could you repeat https://github.com/platformio/platform-atmelavr/issues/55#issuecomment-310778037 ?

See https://github.com/platformio/platformio-core/commit/05043c0f4a88e5d629acb64087ea6023d6365f5b

AndreKR, 2017-06-23:
Doesn't work.

I don't speak Python but I think you forgot to extract the port name from `before`, it seems to compare the ListPortInfo object with the port name ("COM13").

ivankravets, 2017-06-23:
Sorry, you are right! Please repeat https://github.com/platformio/platform-atmelavr/issues/55#issuecomment-310778037

AndreKR, 2017-06-23:
Works! :)

ivankravets, 2017-06-23:
Now, please run `pio upgrade`, it should update to 3.4.0rc1. Does it work?

closed 2017-06-23 by ivankravets

PS-Memis, 2020-10-23:
Hello, 
I have a similar Issue but with an Arduino Leonardo. 
I m trying to upload my code to an arduino leonardo using Vscode + Platformio IDE.
as I connect the mC its been detected in port COM 10.  Uploading the code it tries to force the port for the upload but than it rejected and the shown faillure appears on another port ( COM 11 here for the case). the weird thing is that, on other Laptops, the upload is running normally and the mC is working. I tried in other ports on my laptop and other devices and other mCs to check if the usb is working well and its the case. I have the following error as output :

///////////////////////////////////////
avrdude: ser_open(): can't open device "\.\COM11": The specified file can not be found

avrdude done. Thank you.

*** [upload] Error 1
== [FAILED] Took 23.53
////////////////////////////

I ll be very greatfull if someone can help.

BR. SA_AB