platformio/platform-atmelavr #56: Add support for Arduino UNO WiFi

Vincent-Pang, 2017-06-25:
Hi, please add support for [Arduino UNO WiFi](http://www.arduino.org/products/boards/arduino-uno-wifi).

I am doing some development on this board with CLion. And I would like to use the [Arduino WiFi Link Library](http://www.arduino.org/learning/reference/wifilink).

I found that there are some unowifi related code on [platformio github](https://github.com/platformio/platformio-pkg-framework-arduinoavr/blob/master/boards.txt) (at the end of boards.txt) already, but I can't found unowifi at [board manager](http://platformio.org/boards). Then I install the most similar board uno instead.

The followings are my steps.

pio init --ide clion --board uno
pio lib install "WiFi Link"

main.cpp
```
#include <Arduino.h>
#include <WiFiLink.h>

void setup()
{
    Serial.begin(9600);

    //Check if communication with the wifi module has been established
    if (WiFi.status() == WL_NO_WIFI_MODULE_COMM)
    {
        Serial.println(F("Communication with WiFi module not established."));
        while (true) // don't continue:
        {
        }
    }

}

void loop()
{
}
```

When I build it, compile error occur.
```
Compiling .pioenvs/uno/lib/WiFi Link_ID1541/utility/comm.o
Compiling .pioenvs/uno/lib/WiFi Link_ID1541/utility/server_drv.o
In file included from .piolibdeps/WiFi Link_ID1541/src/utility/comm.cpp:27:0:
.piolibdeps/WiFi Link_ID1541/src/utility/comm.h:40:8: error: 'CommDrv' does not name a type
extern CommDrv commDrv;
^
```

Then I took a look at comm.h. And from this [src code](https://github.com/arduino-org/arduino-firmware-wifilink/blob/master/ArduinoFirmwareEsp/config.h), I think I should define ESP_CH_SPI first although I know I should not modify this file by myself. But anyway, let me try if it could work.
```
#ifndef COMM_H
#define COMM_H

//#if defined(__AVR_ATmega32U4__) || defined(ARDUINO_ARCH_STM32F4)
#if defined(ESP_CH_UART)
    #include "utility/uart/uart_drv.h"
	class CommDrv : public UartDrv {};
#elif defined(ESP_CH_SPI)              // <---- I have to define this
    #include "utility/spi/spi_drv.h"
    class CommDrv : public SpiDrv {};
#endif

extern CommDrv commDrv;

#endif

```

So I define it manually.

```
#ifndef COMM_H
#define COMM_H

#define ESP_CH_UART    // <---- I define it manually

//#if defined(__AVR_ATmega32U4__) || defined(ARDUINO_ARCH_STM32F4)
#if defined(ESP_CH_UART)
    #include "utility/uart/uart_drv.h"
	class CommDrv : public UartDrv {};
#elif defined(ESP_CH_SPI)
    #include "utility/spi/spi_drv.h"
    class CommDrv : public SpiDrv {};
#endif

extern CommDrv commDrv;

#endif
```

And then there is another compile error. Cannot find Wire.h
```
Compiling .pioenvs/uno/src/main.o
Compiling .pioenvs/uno/lib/WiFi Link_ID1541/utility/uart/SC16IS750.o
.piolibdeps/WiFi Link_ID1541/src/utility/uart/SC16IS750.cpp:25:18: fatal error: Wire.h: No such file or directory
#include <Wire.h>
^
compilation terminated.
```

I don't know why it can't find Wire.h. Never mind, let me include it manually.

```
#include <Arduino.h>
#include <Wire.h>   // <---- I include it manually
#include <WiFiLink.h>

void setup() {...}
void loop() {...}
```

Linking error occur this time. I have no idea how to fix it ...
```
Archiving .pioenvs/uno/lib/libWiFi Link_ID1541.a
Indexing .pioenvs/uno/lib/libWiFi Link_ID1541.a
Linking .pioenvs/uno/firmware.elf
/var/folders/s8/g4549n1n2610wz5_5kgbrc540000gn/T//ccmkEqiR.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_main.o.1959':
ccmkEqiR.ltrans0.o:(.text.startup+0x28): undefined reference to `UartDrv::begin()'
/var/folders/s8/g4549n1n2610wz5_5kgbrc540000gn/T//ccmkEqiR.ltrans0.ltrans.o: In function `main':
ccmkEqiR.ltrans0.o:(.text.startup+0x1a4): undefined reference to `UartDrv::waitForSlaveReady()'
ccmkEqiR.ltrans0.o:(.text.startup+0x1a8): undefined reference to `UartDrv::commSlaveSelect()'
ccmkEqiR.ltrans0.o:(.text.startup+0x1b0): undefined reference to `UartDrv::sendCmd(unsigned char, unsigned char)'
ccmkEqiR.ltrans0.o:(.text.startup+0x1b4): undefined reference to `UartDrv::waitForSlaveReady()'
ccmkEqiR.ltrans0.o:(.text.startup+0x1ce): undefined reference to `UartDrv::waitResponseCmd(unsigned char, unsigned char, unsigned char*, unsigned char*)'
ccmkEqiR.ltrans0.o:(.text.startup+0x1d2): undefined reference to `UartDrv::commSlaveDeselect()'
collect2: error: ld returned 1 exit status
*** [.pioenvs/uno/firmware.elf] Error 1
```

Please help to support Arduino UNO WiFi in a proper way.
Thanks.

EngineerHebron, 2018-02-02:
I am having the same problem here!please any one to help

anton-mir, 2020-10-21:
Plz add it

MCUdude, 2020-11-27:
I don't see much need for this as the [Original Uno Wifi](https://store.arduino.cc/arduino-uno-wifi) is dead, and we already have support for its successor, the Arduino UNO Wifi Rev2, which is supported by PlatformIO.