platformio/platform-atmelavr #58: toolchain-atmelavr package for freebsd_amd64

rubyconn, 2017-07-23:
Hello,

I have compiled toolchain-atmelavr on FreeBSD 10.3R amd64 to fix error below. 

```
PackageManager: Installing toolchain-atmelavr @ ~1.40902.0
Error: Could not find a version that satisfies the requirement '~1.40902.0' for your system 'freebsd_amd64'
```

I used the source files from here: http://distribute.atmel.no/tools/opensource/Atmel-AVR-GNU-Toolchain/3.5.4/

I have added a package.json file (edited for correct system) and removed doc, info, man and share directories, and replaced identical files with symlinks, to save space and make the file structure identical with this linux package. 
https://bintray.com/platformio/dl-packages/download_file?file_path=toolchain-atmelavr-linux_x86_64-1.40902.0.tar.gz

Here's the link to the archive.
https://www.dropbox.com/s/x8ovaohyurvl7hr/toolchain-atmelavr-freebsd_amd64-1.40902.0.tar.gz?dl=0

rubyconn, 2017-07-23:
I also compiled the other dependencies for atmelavr platform on freebsd_amd64. Let me know if anything's wrong.

https://www.dropbox.com/s/0yacfcc9p1u9120/tool-avrdude-freebsd_amd64-1.60300.2.tar.gz?dl=0
https://www.dropbox.com/s/gpcqp3u8gol6los/tool-micronucleus-freebsd_amd64-1.200.0.tar.gz?dl=0

ivankravets, 2017-07-23:
That is great work!!! Thank you very much! 👍 

All packages have been deployed to our CDN networks.

Would be thankful for other FreeBSD packages which we don't have.

P.S: Please re-test them. Remove your existing and `pio run` command should download them from our package registry.

closed 2017-07-23 by ivankravets

rubyconn, 2017-07-23:
Thanks for the quick deployment. [=

I tested with `pio run` and it works great.

trombik, 2020-06-19:
for those of you who look for `toolchain-atmelavr` for FreeBSD, you should use https://github.com/trombik/platformio-freebsd-toolchain-atmelavr instead of binaries from unknown sources.