platformio/platform-atmelavr #61: Problem with building usbdrvasm.o in DigisparkKeyboard on digispark-tiny...

brainwagon, 2017-07-28:
While trying to build the example framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard for the digispark-tiny, I get the following.  It appears that this error has previously been reported and fixed....

Any hints?

[Thu Jul 27 21:44:09 2017] Processing digispark-tiny (platform: atmelavr; board: digispark-tiny; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting spark.ino
Collected 69 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <DigisparkKeyboard>

*** Multiple ways to build the same target were specified for: /home/markv/spark/.pioenvs/digispark-tiny/lib/DigisparkKeyboard/usbdrvasm.o  (from ['/home/markv/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/usbdrvasm.S'] and from ['/home/markv/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/usbdrvasm.asm'])
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 273, in BuildLibrary

m0n1s, 2017-08-05:
Got the same error on Visual Studio Code on macOS 10.12.5

asteriskie, 2017-11-10:
Same issue on linux,  both arm and x86_64, doesn't seem to be an architecture issue.  Both fresh installs from platformio

ivankravets, 2017-11-26:
Fixed in https://github.com/platformio/platformio-pkg-framework-arduinoavr/commit/0ca5066ea859618ea8cf11768781d710343f1e08

Will be released soon.

P.S: You can remove `usbdrvasm.asm` manually from `~/.platformio/packages/framework-arduinoavr/libraries/cores/digispark/DigisparkKeyboard`

closed 2017-11-26 by ivankravets