platformio/platform-atmelavr #64: Please add support for Elektor Uno R4 with ATmega328PB

elektor-labs, 2017-08-16:
https://www.elektormagazine.com/labs/elektorino-uno-r4-150790
https://www.elektor.com/elektor-uno-r4

Name: Elektor Uno R4
Platform: Atmel AVR
Frameworks: Arduino
MCU: ATmega328PB
Debug:
Frequency: 16 MHz
RAM: 2 kB
Flash: 31.5 kB

ivankravets, 2017-08-16:
Could you try this configuration file `platformio.ini`?

```
[env:uno]
platform = atmelavr
framework = arduino
board = uno
board_mcu = atmega328pb
```

Does it work? If yes, we will add a new board using Arduino Uno manifest.

Also, do we need new variant?

elektor-labs, 2017-08-17:
I get a linker error:

`Linking .pioenvs\uno\firmware.elf
collect2.exe: error: ld returned 5 exit status
*** [.pioenvs\uno\firmware.elf] Error 1
 [ERROR] Took 8.62 seconds`

Please note that the 328pb requires its own header file, startup code and library:

- iom328pb.h
- crtatmega328pb.o
- libatmega328pb.a

These files are present in toolchain_atmelavr together with "specs-atmega328pb", which is good.

Because the 328pb has 2x UART, 2x SPI, 2x I2C and some more, the Elektor Uno R4 is not a simple Uno variant. An Arduino boards package for the Elektor Uno R4 is available at https://github.com/ElektorLabs/Arduino/releases/download/v1.0.0/package_elektor_uno_r4_1_8_x_index.json
This package adds libraries to support the extra peripherals. It uses tricks to get the files in the right places and switches avrdude into silent mode to avoid warnings.

Furthermore, allthough avrdude may not support the 328pb, but it can easily be added to its config file.

The Atmel ATmega328PB board that is in the PIO boards list does not support its extra peripherals through Arduino and so its support files are probably worthless.

Skysurfer-14, 2017-11-10:
I get the same error, when I use the board "Atmel ATmega328PB".

It would be very helpful to add the support of ATmega328PB and its variants (Elector UNO R4, Wattuino Pro Mini PB, etc.) !

ivankravets, 2017-11-27:
Toolchain is fixed. Please run `pio update` and build project again. If it works, I'll add new board `Elektor Uno R4` with its variant.

ivankravets, 2017-11-27:
@elektor-labs I added your PINs definition https://github.com/platformio/platformio-pkg-framework-arduinoavr/commit/c6fd54b7af58c88519741961adae7943c2f22a06

Please note that I commented `https://github.com/platformio/platformio-pkg-framework-arduinoavr/blob/master/variants/R4/pins_arduino.h#L30`, it produces a lot of errors.

Also, new board manifest was added https://github.com/platformio/platform-atmelavr/commit/c2309b8050c34c72a7d0e33f4eeff715561da6c7

Please re-test. If it doesn't work, please reopen this issue.

Skysurfer-14, 2017-11-27:
Thank you! I've made some first tests after `pio update` and inserting the boards-file and pins_arduino.h in directory R4. 
- `Serial.begin(9600);` and `Serial1.begin(9600);` compiles
- `#include <SPI.h>` and `#include <SPI1.h>` results in compiling errors
- `#include <Wire.h>` and `#include <Wire1.h>` results in compiling errors

ivankravets, 2017-11-27:
I've just deployed new package. So, you can experiment on it via 
```
[env:elektor_uno_r4]
platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino
board = elektor_uno_r4
```

Let's wait for the comment by @elektor-labs

arilow, 2024-10-15:
Hi! Are there any updates on this? I'm having the same issue 

- Serial.begin(9600); and Serial1.begin(9600); compiles
- #include <SPI.h> and #include <SPI1.h> results in compiling errors
- #include <Wire.h> and #include <Wire1.h> results in compiling errors