platformio/platform-atmelavr #67: Arduino Upload Fails

AmanMohanty97, 2017-09-20:
Configuration:
Operating system: Ubuntu 16.04 LTS

PlatformIO version: 
PlatformIO, version 3.5.0a13

Description of problem:
Uploading of a sketch using upload in platformIO doesn't work.

Steps to Reproduce:
platformio.ini is

[env:uno]
platform = atmelavr
lib_extra_dirs = ~/Documents/Arduino/libraries
board = uno
framework = arduino

Actual Results:

[Wed Sep 20 22:34:19 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino; lib_extra_dirs: ~/Documents/Arduino
/libraries)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting Blink.ino
Collected 24 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs/uno/src/Blink.ino.o
Linking .pioenvs/uno/firmware.elf
Checking program size
text       data     bss     dec     hex filename
940           0       9     949     3b5 .pioenvs/uno/firmware.elf
Building .pioenvs/uno/firmware.hex
Looking for upload port...
Auto-detected: /dev/ttyACM2
Uploading .pioenvs/uno/firmware.hex
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

And it goes on for 10 times.

ivankravets, 2017-11-27:
Please install https://github.com/platformio/platformio-core/blob/develop/scripts/99-platformio-udev.rules and check your USB cable

closed 2017-11-27 by ivankravets

JoaquinEduardoArreguez, 2018-03-15:
Hi @ivankravets, I have the same problem issued here and couldn`t solve it using your suggestion, udev rules already installed and USB cable ok, any other ideas on solving this issue?, my info:

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
[Thu Mar 15 14:42:54 2018] Processing atmega328pb (platform: atmelavr; board: atmega328pb; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Atmel ATmega328PB
SYSTEM: ATMEGA328PB 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph

|-- <modulo>
|-- <controlador>
|   |-- <modulo>
Linking .pioenvs/atmega328pb/firmware.elf
Checking program size
text       data     bss     dec     hex filename
1824        158     157    2139     85b .pioenvs/atmega328pb/firmware.elf
Building .pioenvs/atmega328pb/firmware.hex
Configuring upload protocol...

AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pioenvs/atmega328pb/firmware.hex
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00