platformio/platform-atmelavr #68: C++ Standard Library not included in sparkfun_promicro16 env

pyetras, 2017-09-29:
**Operating system**:
OS X
**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.5.0a16
### Description of problem

#### Steps to Reproduce
```
$ platformio run -v
[Thu Sep 28 23:27:53 2017] Processing sparkfun_promicro16 (platform: atmelavr; build_flags: -std=c++11; board: sparkfun_promicro16; framework: arduino)
------------------------------------------------------------------------------------
Collected 25 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Servo> v1.1.2 (/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/Servo)
|-- <Wire> v1.0 (/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire)
|-- <SoftwareSerial> v1.0 (/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial)
avr-g++ -o .pioenvs/sparkfun_promicro16/src/DriveMessage.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -std=c++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_PROMICRO16 -DARDUINO=10619 -DUSB_VID=0x1B4F -DUSB_PID=0x9203 "-DUSB_PRODUCT=\"SparkFun Pro Micro\"" -DUSB_MANUFACTURER=\"SparkFun\" -I/Users/Pietras/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/Pietras/.platformio/packages/framework-arduinoavr/variants/sparkfun_promicro -Isrc -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/Servo/src -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial/src src/DriveMessage.cpp
avr-g++ -o .pioenvs/sparkfun_promicro16/src/main.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -std=c++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_PROMICRO16 -DARDUINO=10619 -DUSB_VID=0x1B4F -DUSB_PID=0x9203 "-DUSB_PRODUCT=\"SparkFun Pro Micro\"" -DUSB_MANUFACTURER=\"SparkFun\" -I/Users/Pietras/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/Pietras/.platformio/packages/framework-arduinoavr/variants/sparkfun_promicro -Isrc -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/Servo/src -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src -I/Users/Pietras/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial/src src/main.cpp
avr-gcc-ar rc .pioenvs/sparkfun_promicro16/libFrameworkArduinoVariant.a
src/DriveMessage.cpp:10:21: fatal error: algorithm: No such file or directory
#include <algorithm>
^
compilation terminated.
```
Also the IDE does not recognize the header statement, std namespace etc.

**The content of `platformio.ini`:**
```ini
[env:sparkfun_promicro16]
platform = atmelavr
board = sparkfun_promicro16
framework = arduino
build_flags = -std=c++11
```

ladislas, 2017-11-08:
the standard library is not compatible with AVR mcu.

some implementations have been made, like [µSTL](https://github.com/msharov/ustl). I have not tried it myself so I can't say if it's great but it seems to be well maintained.

ivankravets, 2017-11-27:
See http://www.atmel.com/webdoc/avrlibcreferencemanual/ch20.html

closed 2017-11-27 by ivankravets