platformio/platform-atmelavr #71: Manchester lib fails on ATTiny

surdu, 2017-11-03:
Following [this tutorial](http://www.instructables.com/id/Attiny85-RF-Transmitter-to-Arduino-Uno-Receiver-Ma/) but using Platform.io instead of arduino, make the radio part not work for the RF receiver (at least).

Programming the receiver with Arduino instead Platform.io makes it work again.

My suspicion is that in Platform.io some definitions are wrong and the Manchester lib being very dependent of a lot of [compile](https://github.com/mchr3k/arduino-libs-manchester/blob/master/Manchester.cpp#L77) [time](https://github.com/mchr3k/arduino-libs-manchester/blob/master/Manchester.cpp#L289) [flags](https://github.com/mchr3k/arduino-libs-manchester/blob/master/Manchester.cpp#L364), does not configure properly. 

Unfortunately I don't have the skills to debug this issue further but I can help with any information one would need to fix this.

ivankravets, 2017-11-27:
Could you provide source code with `platformio.ini` to reproduce this issue?

surdu, 2017-11-27:
Unfortunately, platform IO went into an "installing/restarting" loop, so I had to uninstall it. 

The code in the linked tutorial should do the trick.

ivankravets, 2017-11-27:
> Unfortunately, PlatformIO went into an "installing/restarting" loop, so I had to uninstall it.

This issue was fixed in related Python's Arrow library. Please try VSCode for PlatformIO and provide a project to reproduce the issue.

closed 2017-11-28 by surdu