platformio/platform-atmelavr #72: Provide a `toolchain_dir` env variable for custom toolchains

ladislas, 2017-11-08:
Hi there! 

I'm one of the maintainer of [osx-cross/avr](https://github.com/osx-cross/homebrew-avr), a brew formula to build the avr toolchain with the latest gcc, binutils and avrlibc.

For some year I was using the Sudar's Arduino-Makefile but I've been playing a lot recently with platformio and it's quite amazing!

I'm wondering if you'd be interested in a PR to provide users with a way to set the path to their toolchain. For example, when using the osx-cross/avr, the path is `/usr/local/bin`

If that's something you think might be valuable, I can work on a PR.

Thanks,
Ladislas

ivankravets, 2017-11-21:
Hi,

You can remove all files excluding `package.json` from `~/.platformio/packages/toolchain-atmelavr`. After that, just make symbolic links to `~/.platformio/packages/toolchain-atmelavr/bin`. Here is a list with binary names https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L89

P.S: If you need a custom binary names, toolchain... You can fork with dev/platform and tune all moments.

ladislas, 2017-11-22:
Oh I understand! I thought that just changing the path [here](https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L90) with an external argument would make the job but after some testing I found out that it relies heavily on `toolchain-atmelavr` even if the compilation was done with my own toolchain.

Symlinking is adding one more step and may cause issues in the long run. I'll just fork the repo and create my own platform.

thanks for the help!

closed 2017-11-22 by ladislas

ladislas, 2017-11-26:
I ended up forking the repo and removing the toolchain to provide my own.

https://github.com/osx-cross/platform-osxcrossavr