platformio/platform-atmelavr #76: avrdude -D option often incorrect

seebs, 2017-12-03:
Had a lot of weird problems with avrdude failing when invoked by platformIO, working fine from normal command line. Eventually traced it to the -D option. According to avrdude's documentation:

>  When the -U option with flash memory is specified, avrdude will perform a chip erase before starting any of the programming operations, since it generally is a mistake to program the flash without performing an erase first. This option disables that. Auto erase is not used for ATxmega devices as these devices can use page erase before writing each page so no explicit chip erase is required.

And indeed, that's exactly what I see: Writing to anything that isn't an ATxmega, most changes fail verification and result in a non-working flash state, because the flash needed to be erased. Dropping the "-D" from "upload" seems to fix this.

I do not know how or why this differs from reported behavior in the Arduino IDE, but empirically, when I use that and flash things, it's not using -D.

pfeerick, 2019-06-27:
Very strange, as this is the [same flag the Arduino IDE uses for normal uploads](https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L105), suggesting there's something different in the avrdude conf or the behaviour of the binary. Or maybe this is no longer an issue?

seebs, 2019-06-27:
Huh, I don't know. So far as I know, they were both using the same avrdude. The command lines I got that appeared to work, which I got by using the Arduino IDE, didn't have the -D in them. I have no idea why.