platformio/platform-atmelavr #83: How about support for the ATtiny x14/x17/x18 series?

toddzebert, 2018-01-29:


JonSilver, 2019-07-27:
I'm keen to seen support added for these very capable chips, particularly the attiny1617 and attiny817

SpenceKonde, 2019-07-27:
> Is there any chance of integrating the excellent megaTinyCore from
> @SpenceKonde <https://github.com/SpenceKonde>?

I would love to see my cores (megaTibyCore and ATTinyCore) supported by
PlatformIO, however I'm not familiar with it (I just use the arduino IDE).
What can I do to help this happen?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

stonehippo, 2019-09-05:
@SpenceKonde ATTinyCore is already included in PlatformIO, it seems, though it looks like it the somewhat old 1.2.2 version.

As I understand it, adding megaTinyCore would require it to be included in `framework-arduinoavr`, which lives here: https://github.com/platformio/platformio-pkg-framework-arduinoavr. ATTinyCore is already there under the `core[tiny|tinymodern]` cores. I believe it would also require making some changes adding a new `boards_megatinycore.txt` with board definitions that can be exposed to developers. Did I miss anything, @ivankravets?

SpenceKonde, 2019-09-05:
Oh god, 1.2.2? Yeah, we should try to get ATTinyCore bumped up to 1.3.2 - I've changed and improved a lot since then.... 

Quite frankly, I know very little about PlatformIO, other than that people keep complaining that my core isn't supported there, or that bugs I fixed eons ago in ATTinyCore are still present there. I don't remember who set up ATTinyCore 1.2.2 on PlatformIO, but I know it wasn't me.

stonehippo, 2019-09-05:
@SpenceKonde looking at the commit, @ivankravets was the last one to update the ATTinyCore in PIO. I can take a look at making that update over in the framework repo and submit a PR to get it moved up to 1.3.2.

stonehippo, 2019-09-05:
Ok, I’ve made the pull request, following the pattern of the last update and using ATTinyCore 1.3.2 as the basis. @ivankravets can you point me to the best way to advise folks to test the PR with those updates?

SpenceKonde, 2019-09-05:
Wow! Thank you! <3

So now its just megaTinyCore

ivankravets, 2019-09-05:
Thanks for your PR! Everything is merged and deployed. Please re-test with upstream version of dev/platform http://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

As for the megaTinyCore, @stonehippo you can do similar PR. I would like to ask to use a new board ID notation. We plan to add soon support for starting a project from bare MCU/Device and from Development Kits. Dev/kits can have own name but device ID should have official IF provided by the vendor. Mostly, they are uppercased. For example, instead of `attiny817.json` we should use `ATtiny817.json` => https://www.microchip.com/wwwproducts/en/ATTINY817

stonehippo, 2019-09-05:
Thanks for merging the PR, @ivankravets. I’ll do another for #160 as soon as I get a chance. Will also look at the new board ID notation, too.

stonehippo, 2019-09-07:
@SpenceKonde @ivankravets I’m looking at bring megaTinyCore over, but I noticed a couple of things that lead to some questions that might need to be addressed before the BSP will work on PlatformIO:

1. @SpenceKonde I noticed in your release notes that prior to release 1.0.1, the official Arduino megaAVR BSP needed to be installed to get compiler support, but that changed with 1.0.2+. Can you give me a hint on why that is (e.g. do you now get the compiler installed alongside megeTinyCore automatically?)
2. Also, it looks like upload/flashing gets done via UDPI rather than ISP?
3. @ivankravets if I needed to get the megaAVR compiler installed as a dependency for these MCUs/megaTinyCore, I assume we’d need a new toolchain deployed (e.g. `toolchain-atmelmegaavr`) up on sourceware?

Based on all of this, I’m wondering if it doesn’t make more sense to create a new `platformio-atmelmegaavr` repo if the tools architecture are that different.

JonSilver, 2019-09-07:
> Also, it looks like upload/flashing gets done via UDPI rather than ISP?

Yes this series of AVRs requires UPDI and has no support for ISP.

stonehippo, 2019-09-07:
Thanks, @JonSilver. Looks like I can build a [UPDI programmer from one of the Pro Mini’s I have laying around](https://github.com/SpenceKonde/megaTinyCore/blob/master/MakeUPDIProgrammer.md).

I’m going to have to get a couple of these new MCUs so I can test anything I do. Have you got the breakouts for this up on Tindie, @SpenceKonde? I’m also going to get an Arduino Nano Every or Uno WiFi rev2, since they also use the newer megaAVR arch.

stonehippo, 2019-09-07:
Ok, got a couple of boards from @SpenceKonde. I’ll pick this up again when they arrive, I guess.

stonehippo, 2019-09-07:
@ivankravets FYI I tested the upstream update for ATTinyCore with an ATtiny85 and things are looking good with 4.2.0.

SpenceKonde, 2019-09-07:
megaTinyCore (and the other megaavr parts) requires 7.3.0 of the compiler; the recent versions of ATTinyCore handle this when installed via board manager (but not when installed manually - I'm not quite sure how to do that in a graceful way). I'm not sure how such custom tools are handled in platform IO - you can see the packages that get installed in the board manager json https://github.com/SpenceKonde/ReleaseScripts/blob/master/package_drazzy.com_index.json

stonehippo, 2019-09-16:
Got the boards today (thanks, @SpenceKonde!). This next week is nuts, so I’ll most likely start to work on the megaTinyCore support next week.

stif, 2020-02-04:
Hi Guys, any news on this? I am also interested in the new ATTinys but would love to keep on working with platformIO!
Cheers, Stif

stonehippo, 2020-02-04:
Hi @stif, I've been wrapped in some other work, but have been planning to get back to adding megaTinyCore. I'll take a whack at it this weekend.

iosengineer, 2020-02-05:
Ditto from me. I guess I'll be in the ArduinoIDE for the moment... (and wrecking my nice folder structure).

iosengineer, 2020-02-05:
Ha! Workaround (for the yuck folder structure thing): added a folder in my repo for the Arduino IDE and symbolic-linked the main.cpp to the folder name .ino, as well as linking the "include" files. Doesn't work for all, but it's a simple project.

jamesmyatt, 2020-02-05:
Thanks @stonehippo ! Much appreciated

stonehippo, 2020-02-08:
Quick update: I’m working through adding the board definitions, and will need to make sure the right versions of AVRDUDE and the compiler are available. Will also need to confirm UPDI with `pio run -t upload`. Probably won’t have it all ready for a PR this weekend, but within the next few days.🤞🏾

iosengineer, 2020-02-09:
Hey that’s awesome. Thanks for the effort!

Thanks,
Adam

ghoeffner, 2020-04-16:
I'm also interested in seeing ATtiny1617 support.

SpenceKonde, 2020-04-16:
Just as an FYI (which may be unwelcome), 2.0.1 is basically out (it's in the github now) and like, there are a bunch of things different in it that might be relevant to this effort - basically it's a release to fix all the bad architecture decisions I made before, before things got any more widely deployed than they already were

stonehippo, 2020-04-16:
Thanks @SpenceKonde. I’ve been following 2.x and will bring it in.

SimonMerrett, 2020-05-17:
@stonehippo thanks for all your work on this. I'm transitioning to pio from Arduino IDE now and asked @SpenceKonde if he had any appetite to support pio. He pointed me here and after reading back through the issue, googling around pio community and looking at issues related to @MCUdude's MiniCore, MegaCore and MightyCore + documentation on setting fuses, I'm none-the-wiser about a) has support for megaTinyCore been added and b) is there an efficient way of importing necessary files without having to handwrite the platformio.ini for every combination of IC and fuse setting?
Thanks for anyone who can enlighten me.

stonehippo, 2020-05-17:
@SimonMerrett short answer is that I got pulled into a bunch of other stuff and haven't had time to finish integrating MegaTinyCore into PIO yet. Good news is I'm taking a few days off, so I hope to get to it this week.

SimonMerrett, 2020-05-17:
Thanks @stonehippo , hopefully I'll have worked out how to use pio a bit better by then.

SimonMerrett, 2020-05-21:
@stonehippo is there any community contribution that could help with the megaTinyCore integration?

MCUdude, 2020-06-03:
> He pointed me here and after reading back through the issue, googling around pio community and looking at issues related to @MCUdude's MiniCore, MegaCore and MightyCore + documentation on setting [...]

The first step would be to add megaTinyCore, where you'll have to manually write the fuses in platformio.ini if you have to set the fuses. The next step would be to create a new fuses.py/bootloader.py script similar to what's currently used to set fuses + burn bootloader for  Mighty/Mega/Mini/Major/MicroCore. It's a tedious job for sure, but it's definitely doable. What's awesome is that the tiny0's, tiny1's, mega0's and AVR-DA's have more or less the same fuse settings. This means that with a little cross-referencing it's possible to create a script that can calculate the fuses for more than 30 new targets!

And the script and platformio.ini don't have to include absolutely all possible settings. We just need the most common ones, just like the current script for "classic" AVRs.

SimonMerrett, 2020-06-03:
> The first step would be to add megaTinyCore, where you'll have to manually write the fuses in platformio.ini if you have to set the fuses. The next step would be to create a new [...]

@MCUdude that's great info, thanks. Do you have any idea how the need for UPDI and the common reliance on the avr-based 'jtag2updi' hardware programmers will affect this port?

MCUdude, 2020-06-03:
> Do you have any idea how the need for UPDI and the common reliance on the avr-based 'jtag2updi' hardware programmers will affect this port?

Well, the programmer has to be specified in platformio.ini so that avrdude can utilize it. As long as the Avrdude version PlatformIO uses is up to date, you can use whatever UPDI-compatible programmer you like. It shouldn't be too much work I think.

SimonMerrett, 2020-06-03:
@MCUdude thanks again. That will be helpful, I'm sure.
@ivankravets thanks for assigning someone to this. 
@valeros please say if this needs community help and we will try to support.

SimonMerrett, 2020-06-29:
@valeros I know almost nothing about PIO so I was hopeful when @ivankravets assigned you that someone who knew PIO would be able to do something with this. But if there are non-PIO-internal elements that the community can help with, please explain what we need to do.

lou270, 2020-11-15:
I'm very interested by this support in PIO, any news about it ?

MCUdude, 2020-11-15:
this issue Doesn't really belong here, it belongs in the [platform-atmelmegaavr](https://github.com/platformio/platform-atmelmegaavr) repo.

* Someone would have to make all the manifest files. 25 of them if all [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) targets should be supported
* Someone should figure out all the fuse settings and add them to the fuses.py script so they can be calculated dynamically.

There is also a lot of build settings in the megaTinyCore boards.txt file that needs to be figured out like setup time, millis timer, BOD modes, baud rate voltage, etc. We would probably get the best PlatformIO implementation if @SpenceKonde himself worked together with the PlatformIO developers to figure out how to implement all the quirks and features, similar to what [I did with MegaCoreX](https://github.com/platformio/platform-atmelmegaavr/issues/2) a few weeks back

stonehippo, 2020-11-17:
Looks like @valeros has already started on adding megaTinyCore on **platform-atmelmegaavr**. If that’s where it’ll live, it seems like this one should get closed out, though ATTinyCore could use another update since it hadn’t been updated since the 1.3.2 PR a while back.

But that’s another issue, right, @ivankravets?