platformio/platform-atmelavr #90: Upload to arduino UNO fails

JoaquinEduardoArreguez, 2018-03-15:
Hi everyone, I am trying to upload to UNO board but avrdude keeps saying that is out of sinc with my UNO board,

udev rules already installed and USB cable ok, any other ideas on solving this issue?, my info:

Ubuntu 16.04, platformio on Atom, permissions to use the upload port OK.

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
[Thu Mar 15 14:42:54 2018] Processing atmega328pb (platform: atmelavr; board: atmega328pb; framework: arduino)
Verbose mode can be enabled via -v, --verbose option
PLATFORM: Atmel AVR > Atmel ATmega328PB
SYSTEM: ATMEGA328PB 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph

|-- 
|-- 
| |-- 
Linking .pioenvs/atmega328pb/firmware.elf
Checking program size
text data bss dec hex filename
1824 158 157 2139 85b .pioenvs/atmega328pb/firmware.elf
Building .pioenvs/atmega328pb/firmware.hex
Configuring upload protocol...

AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pioenvs/atmega328pb/firmware.hex
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

kodaman2, 2018-03-31:
Same on OSX Sierra, trying out the simple blink example.
Arduino Uno, I can see TX, RX blinking so I know communications are ok, just can't upload the blink example at all. Works on Arduino IDE.
```
[Sat Mar 31 16:10:56 2018] Processing atmega328pb (platform: atmelavr; board: atmega328pb; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Atmel ATmega328PB
SYSTEM: ATMEGA328PB 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)

Collected 24 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/atmega328pb/src/Blink.cpp.o
Archiving .pioenvs/atmega328pb/libFrameworkArduinoVariant.a
Indexing .pioenvs/atmega328pb/libFrameworkArduinoVariant.a
Compiling .pioenvs/atmega328pb/FrameworkArduino/CDC.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/HardwareSerial0.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/HardwareSerial1.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/HardwareSerial2.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/HardwareSerial3.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/PluggableUSB.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/Stream.cpp.o
ompiling .pioenvs/atmega328pb/FrameworkArduino/Tone.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/USBCore.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/WInterrupts.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/WMath.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/WString.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/_wiring_pulse.S.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/abi.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/hooks.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/main.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/new.cpp.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/wiring.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/wiring_analog.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/wiring_digital.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/atmega328pb/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/atmega328pb/libFrameworkArduino.a
Indexing .pioenvs/atmega328pb/libFrameworkArduino.a
Linking .pioenvs/atmega328pb/firmware.elf
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: /dev/cu.usbmodem1421
Uploading .pioenvs/atmega328pb/firmware.hex
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
```

kodaman2, 2018-03-31:
Ok one thing to note. When importing the blink example to platformio it will not upload to the uno. However when starting a new project and pasting the blink code it works flawlessly. Not sure why.

ivankravets, 2018-04-03:
> Not sure why.

Because you selected incorrect board in `platformio.ini` , see `atmega328pb` in logs

closed 2018-04-03 by ivankravets