platformio/platform-atmelavr #95: PIO Unified Debugger for AVR

ivankravets, 2018-06-04:
Moved from https://github.com/platformio/platformio-core/issues/1645

/cc @suroh001

PLorand, 2018-07-10:
Maybe this can be used as a reference: 
https://www.avrfreaks.net/forum/tutsoftgcc-avr-debugging-using-gdb-and-avr-dragon-under-linux-ubuntu

aster94, 2019-01-03:
hello, have you seen this?
https://github.com/dcwbrown/dwire-debug

vindicatorr, 2019-06-18:
@aster94 I came across that today and was very hopeful of making use of it... buuuuut I was bummed to find that atmega16 doesn't have the capability to use it, only JTAG.
I keep coming across avarice and may look into that more.

cebas, 2019-11-18:
Hi @ivankravets , I would love to be able to debug on my arduino uno. Is there any advance on this feature?
Thanks

jxsl13, 2020-04-06:
Highly requested feature.

BluRaf, 2020-10-07:
It looks like it would be possible to use [AVaRICE](http://avarice.sourceforge.net/) debugger server to interface GDB with JTAG/ICE debugger.
Someone on PlatformIO forums was already trying to use it with mixed results: https://community.platformio.org/t/atmega32-debugging-with-avarice-and-avr-gdb/15960

barrydegraaff, 2020-10-23:
Hello, I would also like to see AVR debugger support in PlatformIO, even though I can debug just fine over the command line.

Attached 2 documents that document how to debug using ONLY open-source software the Atmega 328p and Atmega 2560 using USB AVR JTAGICE MKII-CN:

[debugging-atmega328_README.pdf](https://github.com/platformio/platform-atmelavr/files/5427662/debugging-atmega328_README.pdf)

[debugging-atmega2560_README.pdf](https://github.com/platformio/platform-atmelavr/files/5427664/debugging-atmega2560_README.pdf)

If a developer of PlatformIO is interested, I am willing to ship a USB AVR JTAGICE MKII-CN **FOR FREE** as donation, but only to a developer that can prove (by commits) they are actively involved in the project. Thanks!

N0cynym, 2022-12-21:
@ivankravets any updates about debugging AVR?

maxgerhardt, 2023-01-15:
I'm the CI maintainer of [the AVaRICE program](https://github.com/avrdudes/avarice) now, which is supposed to be able to use several debugging probes (Atmel-ICE, AVR Dragon, mEDBG, eDBG, ...) to open a GDB server. After some testing on real hardware (got an ATMega328P XPlained Mini with mEDBG probe on the way) and packaging of binaries I'll then be excited to bring these improvements into PlatformIO for AVR debugging support.

maxgerhardt, 2023-01-20:
I am making slow but steady progress with AVaRICE on a ATMega328P XPlained Mini (mEDBG) board, incorporating a few patches, recompiling avr-gdb with XML support, and now I got PlatformIO + avarice + avr-gdb to load the firmware binary through debug wire, set a breakpoint and have it stop. Line by line debugging of the chip actually works.

[image]

Excited to test more and make it better.

SunboX, 2025-05-22:
Any progress on this? I was looking for a way to debug a ATTiny88 using PlatformIO and came across here.