platformio/platform-atmelavr #100: macOS will soon drop 32-bit support

tiferrei, 2018-06-21:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [x] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.14 Beta (18A293u)

**PlatformIO Version** (`platformio --version`): 3.5.3

### Description of problem
macOS Mojave drops support for 32 bit binaries, the atmelavr toolchain seems to be almost all 32 bit, which means that every time I compile something I get:
[image]

Here's a list of all the platformio binaries the system has complained about so far:
[image]

More info here: https://support.apple.com/en-gb/HT208436

#### Steps to Reproduce

1. Install Mojave
2. `platformio run`

### Additional info
Everything still works for now, this is more of a deprecation warning.
Is there any way I can fix this?

Thanks,
Tiago

ivankravets, 2019-04-23:
Should be fixed with the latest release.

MacWyznawca, 2019-05-05:
I have the current version (there is no newer version according to platform update) and still macOS issues 32-bit warnings.

MacWyznawca, 2019-05-06:
But my problem is with cc1plus command. It's runing when I complile AVR core (no Arduino).

noisymime, 2019-06-05:
I'm still seeing this with the latest versions installed also. The following utilities (at least) are running 32-bit versions:
`avr-as`
`avr-ld`
`avr-c++`
`collect2`
`cc1plus`
`lto1`

All of these are executed when doing a compile. Additionally the `avr-dude` binary is 32-bit. 
I've also confirmed with the developer preview that compiling and uploading using the currently published tools does not work under macOS 10.15

Versions:
PIO: 3.6.7
atmelavr:                        @ 1.14.0
toolchain-atmelavr:              @ 1.50400.0 
framework-arduinoavr:            @ 4.0.0 
tool-avrdude:                   @ 1.60300.190424

ivankravets, 2019-06-05:
Atmel AVR toolchains (Windows, Linux 32/64, macOS) has been updated to the latest official version by Microchip:
- https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers

Also, patches for Atmega328PB were applied using http://packs.download.atmel.com/

Please update PlatformIO packages via `$ pio update`.

closed 2019-06-05 by ivankravets

noisymime, 2019-06-05:
Thanks @ivankravets, I can confirm all compile tools are working after updating to 1.50400.180928 of `toolchain-atmelavr`

`avr-dude` from the `tool-avrdude` package remains 32-bit though. Does this need a separate issue over at https://github.com/platformio/platform-atmelavr/ ?

ivankravets, 2019-06-05:
Yes, please create a separate issue for avrdude.