platformio/platform-atmelavr #101: Outdated micronucleus tool (mac)

ex-punctis, 2018-06-22:
When I pick digispark-usb as the board, PlatformIO installs micronucleus tool version 1.2 (mac binaries) whereas the current version is >2. I wonder if there is a way to modify some setting so it fetches the latest version instead?

ivankravets, 2018-08-11:
We use the latest compatible 2.0 version. X.Y.Z is internal version naming. See Y as a real version name.

ex-punctis, 2018-08-14:
Thank you for your response. Unfortunately, the current version of the command line tool doesn't work with version 2.03 bootloader from https://github.com/micronucleus/micronucleus

The following error is reported: 

> Warning: device with unknown new version of Micronucleus detected.
> This tool doesn't know how to upload to this new device. Updates may be available.
> Device reports version as: 2.3

Since digispark clones are likely to come with the latest version of micronucleus, it might be worthwhile updating the command line tool to bring support for this version. I believe the command line tool is backwards compatible, so devices with older versions of the bootloader should not be affected.

ivankravets, 2018-08-15:
Does it work with Arduino IDE https://digistump.com/wiki/digispark/tutorials/connecting ?

ex-punctis, 2018-08-16:
Yes, it does work with digispark in Arduino IDE using the package you linked. The tool reports version 2.0a4 (whereas the latest one available directly from 
https://github.com/micronucleus/micronucleus is 2.0a5).

ivankravets, 2018-10-25:
Packages are updated. Please run `pio update`.

 Does it work now?

ex-punctis, 2018-10-25:
@ivankravets 
It works now, thank you.

ivankravets, 2018-10-25:
Happy coding with PlatformIO! 😊

t-oster, 2020-01-22:
I still have this issue. I am using platformio on arch linux within visual studio code. It installs this version: 
```
{
    "description": "Micronucleus",
    "name": "tool-micronucleus",
    "system": [
        "linux_x86_64"
    ],
    "url": "https://github.com/micronucleus/micronucleus",
    "version": "1.200.4"
}%   ```
which is not compatible with the 2.3 bootloader. However if I replace the micronucleus binary with a current version from an arch-linux package, it works.
I did an pio update but it did not update anything.

ansemjo, 2020-03-15:
Yep, this is an issue again. I flashed the `micronucleus` bootloader compiled from current master ([e74ce6f](https://github.com/micronucleus/micronucleus/tree/e74ce6f064e0bcbe1c52459a0988187c76834222/)) to an ATtiny85, which reports as being device version 2.4:

```
Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.4
```

I can't seem to find the repository from which `tool-micronucleus` is installed? Is there a place one could correctly create a pull request against? Either way, @ivankravets please reopen.

This is my `platformio.ini`:

```ini
[env]
framework = arduino
platform = atmelavr

[env:tiny85]
board = attiny85
board_build.f_cpu = 16500000
upload_protocol = micronucleus
```

sanbeg, 2020-04-24:
I'm seeing a similar issue on Linux, it's unable to flash a board which uses the latest released bootloader from https://github.com/micronucleus/micronucleus/releases/tag/2.04

It looks like the platformio micronucleus binary on Linux is a few years out of date, it would be better to update to the latest release, so that it would work with old or new bootloaders.

sanbeg, 2020-04-24:
I've also noticed that the platformio micronucleus crashes if the board is connected when I hit upload; I've never seen that issue on arduino IDE.  Replacing with the latest release resolves that issue, and allow me to upload using a reset button.