platformio/platform-atmelavr #102: ATMega328PB: SPI Library all dead

streamliner18, 2018-06-26:
I'm trying to load the RF24 Example Project into a ATmega328PB board. Turns out it spells havoc all over the place complaining about the absence of register definitions tied to SPI: 
```
#include "RF24.h"
^
compilation terminated.
In file included from C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.cpp:14:0:
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h: In static member function 'static void SPIClass::beginT
ransaction(SPISettings)':
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:203:5: error: 'SPCR' was not declared in this scope

SPCR = settings.spcr;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:204:5: error: 'SPSR' was not declared in this scope
SPSR = settings.spsr;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h: In static member function 'static uint8_t SPIClass::tra
nsfer(uint8_t)':
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:209:5: error: 'SPDR' was not declared in this scope
SPDR = data;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:217:14: error: 'SPSR' was not declared in this scope
while (!(SPSR & _BV(SPIF))) ; // wait
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h: In static member function 'static uint16_t SPIClass::tr
ansfer16(uint16_t)':
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:223:11: error: 'SPCR' was not declared in this scope
if (!(SPCR & _BV(DORD))) {
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:224:7: error: 'SPDR' was not declared in this scope
SPDR = in.msb;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:226:16: error: 'SPSR' was not declared in this scope
while (!(SPSR & _BV(SPIF))) ;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:230:16: error: 'SPSR' was not declared in this scope
while (!(SPSR & _BV(SPIF))) ;
^
C:\Users\wxhem\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h:233:7: error: 'SPDR' was not declared in this scope
SPDR = in.lsb;
```
More errors are omitted. Here's my `platform.ini`:
```
[env:atmega328pb]
platform = atmelavr
board = atmega328pb
framework = arduino
board_build.f_cpu = 8000000L
upload_protocol = usbasp
```

tito, 2018-09-01:
I wanted to try with platformio, and got the same issue here.
To make the chip work with arduino, i needed to change avrdude.conf atmega238p signature from 0x1e 0x95 0x0f to 0x1e 0x95 0x16 (PB signature), then i can select the Arduino Pro Mini 8Mhz 3.3V.
And the compilation works.

With platformio, i've seen that there is a new definition in avrdude for atmega328pb, however, when using `board = pro8MHzatmega328`, i hit the issue:
```
avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: Expected signature for ATmega328P is 1E 95 0F
Double check chip, or use -F to override this check.
```

If i use `board = pro8MHzatmega328`, i hit the same issue as OP.

streamliner18, 2018-09-01:
@tito Changing to a ArduinoPro doesn't solve the problem though. It leverages the fact that 328PB is backward compatible with 328 on the binary level, but will not work if your code is already 328PB-specific.

On your signature problem, DO NOT modify the sig in avrdude.conf, that's too destructive. Simply add `upload_flags = -F`, and it will override the signature check.

tito, 2018-09-09:
Adding -F impact all my others projects, and requires me to be even more cautious when switching from one board to another (i code theses days 2 boards type at the same time, so 2 instances of arduino).
Changing the signatures is for me safer at the moment.

LestherSK, 2018-10-29:
That's because the Atmega328PB has 2 SPI ports, thats why you see 

> 'SPSR' was not declared in this scope

 take a look a this appnote to catch up on the differences. [https://www.pololu.com/file/0J1464/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf](url)

MCUdude, 2019-07-11:
I'm sure this will work if you use ATmega328PB with MiniCore.
[https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md)

ivankravets, 2019-07-11:
Thanks, @MCUdude, great documentation! https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md

We will release this dev/platform tomorrow or on Monday.

closed 2019-07-11 by ivankravets