platformio/platform-atmelavr #103: New arduino Nano boards have new bootloader

ivankravets, 2018-07-02:
PIO Home feedback:

> New arduino Nano boards have new bootloader. Uploading code fails if  software tries to compile for old ATMEGA328P. Same problem if using  Arduino IDE without up to date "Arduino AVR boards"

rcjcooke, 2018-07-16:
Couldn't find an "upvote" mechanism on here so posting instead. Can't seem to upload to a new Nano through PlatformIO on VSCode. Uploading works fine on Arduino IDE. My guess is that it's the new bootloader on the Nano that's the problem. I get the same problems as mentioned here: https://community.platformio.org/t/nano-328p-boot-loader/3997

ivankravets, 2018-08-11:
Please use new board manifest by id `nanoatmega328new` and name `Arduino Nano ATmega328 (New Bootloader)`

commit c6268a22 (2018-08-11): Add new board: Arduino Nano ATmega328 (New Bootloader) // Resolve #103
commit ccbc004b (2018-08-11): Add new board: Arduino Nano ATmega328 (New Bootloader) // Resolve #103
closed 2018-08-11 by ivankravets