platformio/platform-atmelavr #104: argument parsing for avrdude (port and upload speed) incorrect.

sticilface, 2018-07-23:
pio ini file
```
[env:blink]
platform = atmelavr
framework = arduino
board = attiny85
upload_protocol = buspirate
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port =  /dev/cu.usbserial-AI04Y0KK
upload_speed = 115200
```

output verbose
```
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [========= ]  87.9% (used 450 bytes from 512 bytes)
PROGRAM: [======    ]  61.4% (used 5028 bytes from 8192 bytes)
.pioenvs/blink/firmware.elf  :
section                    size      addr
.text                      5020         0
.data                         8   8388704
.bss                        442   8388712
.comment                     17         0
.note.gnu.avr.deviceinfo     60         0
.debug_info                1056         0
.debug_abbrev              1000         0
.debug_line                  26         0
.debug_str                  361         0
Total                      7990
<lambda>(["upload"], [".pioenvs/blink/firmware.hex"])
AVAILABLE: buspirate
CURRENT: upload_protocol = buspirate
BeforeUpload(["upload"], [".pioenvs/blink/firmware.hex"])
avrdude "-P/dev/cu.usbserial-AI04Y0KK -b115200" -v -p attiny85 -C /Users/amelvin/.platformio/packages/tool-avrdude/avrdude.conf -c buspirate -b 115200 -e -D -U flash:w:.pioenvs/blink/firmware.hex:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/Users/amelvin/.platformio/packages/tool-avrdude/avrdude.conf"
User configuration file is "/Users/amelvin/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/cu.usbserial-AI04Y0KK -b115200
Using Programmer              : buspirate
Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "/dev/cu.usbserial-AI04Y0KK -b115200": No such file or directory

avrdude done.  Thank you.

*** [upload] Error 1
```

cause of error `"` incomplete in avrdude command parameters
```
avrdude "-P/dev/cu.usbserial-AI04Y0KK -b115200" -v -p attiny85 -C /Users/amelvin/.platformio/packages/tool-avrdude/avrdude.conf -c buspirate -b 115200 -e -D -U flash:w:.pioenvs/blink/firmware.hex:i
```

removing the `"` results in upload
```
avrdude -P/dev/cu.usbserial-AI04Y0KK -b115200 -v -p attiny85 -C /Users/amelvin/.platformio/packages/tool-avrdude/avrdude.conf -c buspirate -b 115200 -e -D -U flash:w:.pioenvs/blink/firmware.hex:i
```

sticilface, 2018-07-23:
as a temporary fix using this as the upload_flags works :)
```
upload_flags = -P$UPLOAD_PORT" -b"$UPLOAD_SPEED
```

sticilface, 2018-07-23:
or even... the upload speed is optional so just leave 
```
upload_flags = -P$UPLOAD_PORT
```

and it works

sticilface, 2018-07-23:
also...  the pio command uses both flags -D and -e 

```
-D: This disables erasing the chip before programming. We don't want that so don't use this command switch.
-e: This erases the chip, in general we don't use this because we auto-erase the flash before programming.
```
taken from http://www.ladyada.net/learn/avr/avrdude.html

Is this not counter productive? The arduino IDE does not use either of these

justind000, 2018-07-29:
I have also run into an issue with the avrdude command line being invalid. 
In --verbose output, pio sends this command:
`avrdude "-U lfuse:w:0xe2:m -U hfuse:w:0x5f:m -U efuse:w:0xff:m" -v -p attiny85 -C /home/justin/.platformio/packages/tool-avrdude/avrdude.conf -c usbtiny -e -D -U flash:w:.pioenvs/attiny85/firmware.hex:i`

Which complains about in invalid parameter for lfuse.

In my platformio.ini I have:
`upload_flags = -U lfuse:w:0xe2:m -U hfuse:w:0x5f:m -U efuse:w:0xff:m`

If I change the avrdude to this, without the upload_flags in quotes, it works:
`avrdude -U lfuse:w:0xe2:m -U hfuse:w:0x5f:m -U efuse:w:0xff:m -v -p attiny85 -C /home/justin/.platformio/packages/tool-avrdude/avrdude.conf -c usbtiny -e -D -U flash:w:.pioenvs/attiny85/firmware.hex:i`

As of 7/29/2018, I have run pio update/upgrade for the latest versions of everthing. avrdude ver 6.3.

This issue started effecting projects I have had working for over a year and appears to be related to @sticilface identified issue of stray `"`.

ivankravets, 2018-07-29:
See docs http://docs.platformio.org/en/latest/projectconf/section_env_upload.html

Please put each argument in the new line

closed 2018-07-29 by ivankravets

ivankravets, 2018-07-29:
Thanks! Fixed => https://github.com/platformio/platformio-docs/commit/5141b99e54c47ff4057f1e28d95765e3842dc044