platformio/platform-atmelavr #106: ATtiny13A support

maz3max, 2018-08-14:
Hey guys!
I recently did a project on ATtiny13A and discovered that there is only support for ATtiny13.
It was an easy hack to replace the device-specs of the tiny13 with tiny13a inside the avr-toolchain and edit the macro __AVR_ATtiny13__ to __AVR_ATtiny13A__ in c_cpp_properties.json.
The **reason** I need this is that the ATtiny13A has some additional features. (like PRR)
Can you please create another board for that?

ivankravets, 2018-10-10:
> It was an easy hack to replace the device-specs of the tiny13 with tiny13a inside the avr-toolchain

Which changes?

ivankravets, 2019-07-11:
Done in https://github.com/platformio/platform-atmelavr/blob/develop/boards/attiny13a.json

closed 2019-07-11 by ivankravets