platformio/platform-atmelavr #108: Unable to upload code to Arduino Nano (using Arduino IDE works)

AlbertEmil, 2018-09-23:
I am using an Arduino Nano (Clone) which I was programming with the Arduino IDE (Version 1.8.5 on Mac OS X 10.10.5).

Using the Arduino IDE I need to re-plug the Nano before uploading, but after doing this everything works as expected and the sketch is being uploaded successfully.

I then tried to use Platformio core (CLI) and get the well-known `not in sync` error:

    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0
    avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xe0
    avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0
    avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

As said, this error is well-known, esp. in relation to Arduino clones, and reported several times around the Internet with different solutions. 

I just wonder how Arduino IDE is able to upload the code successfully, but Platformio is not since both Arduino IDE and Platformio CLI are just calling `avrdude` to upload the code.

Just to make sure the installation of Platformio is working as desired I switched to an Arduino Uno board, added the additional environment by `platformio init --board uno`, compiled the code and uploaded it which works like a charm (with Arduino IDE as well as Platformio).

I enabled verbose output for both Arduino IDE and Platformio to get the exact `avrdude` command executed to upload the compiled hex-file. I then adopted the call to `avrdude` from the Arduino IDE with the path of the compiled hex-file:
    
    avrdude -v -p atmega328p -c arduino -b 57600 -D -P /dev/cu.usbserial-00000000 -U flash:w:.pioenvs/nano/firmware.hex:i

Using the Arduino IDE's `avrdude` call to upload the hex file which was built by using Platformio works as desired, so I had a look at Platformio's call to `avrdude` which is slightly different form the Arduino one and provides an additional config-file with passing the `-C` option (neglecting possible pre-upload scripts here):

    avrdude -v -p atmega328p -C /Users/Albert/.platformio/packages/tool-avrdude/avrdude.conf -c arduino -b 57600 -D -P "/dev/cu.usbserial-00000000" -U flash:w:.pioenvs/nano/firmware.hex:i

Unfortunately, I am not able to run this command from the Terminal by myself due to an error:

    avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2009 Joerg Wunsch
    
             System wide configuration file is "/Users/Albert/.platformio/packages/tool-avrdude/avrdude.conf"
    avrdude: syntax error at /Users/Albert/.platformio/packages/tool-avrdude/avrdude.conf:1080

Arduino IDE's version of `avrdude` is packed with the Arduino IDE itself and uses a config-file in the `etc` subdirectory of the Arduino application directory. Since I am assuming that the error might come from different config files, I compared both. The only difference in the `ATmega328` section is an additional

```
part parent "m328"
    id          = "m328pb";
    desc        = "ATmega328PB";
    signature       = 0x1e 0x95 0x16;

    ocdrev              = 1;
;
```

in Platformio's config-file.

For the sake of completeness I should mention that I have three different versions of `avrdude` installed on my system:

- shipped with Arduino IDE application,
located at `/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude`,
config file at `/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf`
- shipped with Platformio,
located at `~/.platformio/packages/tool-avrdude/avrdude`,
config file at `~/.platformio/packages/tool-avrdude/avrdude.conf`
- shipped with [CrossPack-AVR,](https://www.obdev.at/products/crosspack/index.html) 
located at `/usr/local/CrossPack-AVR/bin/avrdude` and globally available on Terminal as `avrdude`,
no obvious config file (neither default config file at `/etc/avrdude.conf` nor `~/.avrduderc` available)

ivankravets, 2018-10-10:
Resolved in https://github.com/platformio/platform-atmelavr/commit/ccbc004b4539d137cf9e18072cfe921634ca2ec5

Please use `board = nanoatmega328new` in your `platformio.ini`.

closed 2018-10-10 by ivankravets

AlbertEmil, 2018-10-31:
Neither `nanoatmega328new` nor `nanoatmega328` did do the trick on my Mac. However I was able to upload/flash the code to both of my tested boards with using PlatformIO Core (CLI) on Fedora 29. Due to that, I assume that this issue is not related to the boards itself (from a hardware perspective), but more an avrdude issue of PlatformIO (since using Arduino IDE's version of avrdude, if you can call it that way) is working as expected.