platformio/platform-atmelavr #111: dyld: Library not loaded: @executable_path/libusb-0.1.4.dylib

aosodoev, 2018-10-31:
Getting this error when trying to upload to Digispark Tiny:
```
dyld: Library not loaded: @executable_path/libusb-0.1.4.dylib
Referenced from: /Users/**********/.platformio/packages/tool-micronucleus/micronucleus
Reason: image not found
```
platfomio.ini: 
```
...
[env:control]
platform = atmelavr
board = digispark-tiny
framework = arduino
```

Was able to upload successfully after creating symlinks to libusb-* files:
```
$ cd ~/.platformio/packages/tool-micronucleus/
$ ln -s lib/libusb-0.1.4.dylib
$ ln -s lib/libusb-1.0.0.dylib 
```

leonty, 2019-09-08:
@ivankravets Could you please, take a look at this. I ran into the same problem now in a year after report.

DeviousPenguin, 2019-09-11:
Thanks for posting the workaround with the symlinks, it now seems to upload OK for me on MacOS.