platformio/platform-atmelavr #112: Support for megaAVR 0-Series

sstaub, 2018-11-05:
Can you add support for the new megaAVR 0-series boards, like Arduino Wifi Rev. 2,
ATmega4809-XPRO (ATMEGA4809-XPRO) and ATmega4809 Curiosity Nano Evaluation kit.
The last costs under 10€ inclusive mEDBG https://www.microchip.com/DevelopmentTools/ProductDetails/DM320115
The Arduino core is released at https://github.com/arduino/ArduinoCore-megaavr
I think there is a new platform necessary.

sstaub, 2018-11-28:
Meanwhile the UNO Wifi 2 board from Arduino is released.

sstaub, 2019-03-11:
There is also an interesting variant of the core
https://github.com/MCUdude/MegaCoreX

DesktopMan, 2019-04-16:
Has there been any work on this? The official core for Arduino Uno Wifi Rev2 can be found here:
https://github.com/arduino/ArduinoCore-megaavr

sstaub, 2019-05-18:
Arduino announced a new board  "Arduino Nano Every", also with the new ATMega4809.

sstaub, 2019-07-16:
Hi Ivan, I should give you a ping when finished PIO v4.0

mrarmyant, 2019-07-17:
Is this dead?

sstaub, 2019-07-18:
No, I hope Ivan will support the new AVRs soon.

4ND3R50N, 2019-09-03:
Hello guys, how´s the status?

MCUdude, 2019-09-11:
Since the new megaavr-0 series has to be treated as a completely new architecture it's not as easy as just adding a new AVR board to the list. It requires a fair bit of work and testing. For this, a new platform repo should be made; `platform-megaavr`.

sstaub, 2019-09-11:
I agree.

ivankravets, 2019-10-13:
@valeros see a progress https://community.platformio.org/t/support-for-megaavr-0-series/8064/13