platformio/platform-atmelavr #113: digispark/digistump ATTINY85 board keyboard compile errors

tcurdt, 2018-11-05:
```
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
```
src/keyboard.cpp:
```
#include "DigiKeyboard.h"

void setup() {
  DigiKeyboard.update();
}

void loop() {
  delay(1000);

  DigiKeyboard.update();
  delay(100);

  DigiKeyboard.print("hello");
  DigiKeyboard.println("world");

  delay(5000);
}
```

```
$ platformio run --target upload
Processing digispark-tiny (platform: atmelavr; board: digispark-tiny; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/digispark-tiny.html
PLATFORM: Atmel AVR > Digispark USB
SYSTEM: ATTINY85 16MHz 512B RAM (5.87KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DigisparkKeyboard>
Compiling .pioenvs/digispark-tiny/src/keyboard.cpp.o
Compiling .pioenvs/digispark-tiny/lib379/DigisparkKeyboard/oddebug.c.o
Compiling .pioenvs/digispark-tiny/lib379/DigisparkKeyboard/osccal.c.o
In file included from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Stream.h:24:0,
from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/TinyDebugSerial.h:31,
from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/WProgram.h:18,
from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Arduino.h:4,
from ~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:10,
from src/keyboard.cpp:1:
~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Print.h:37:0: warning: "BIN" redefined
#define BIN 2
^
In file included from ~/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotn85.h:38:0,
from ~/.platformio/packages/toolchain-atmelavr/avr/include/avr/io.h:428,
from ~/.platformio/packages/toolchain-atmelavr/avr/include/avr/interrupt.h:38,
from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/WProgram.h:8,
from ~/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Arduino.h:4,
from ~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:10,
from src/keyboard.cpp:1:
~/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
#define BIN     7
^
In file included from ~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:13:0,
from src/keyboard.cpp:1:
~/.platformio/packages/toolchain-atmelavr/avr/include/avr/delay.h:36:2: warning: #warning "This file has been moved to <util/delay.h>." [-Wcpp]
#warning "This file has been moved to <util/delay.h>."
^
Compiling .pioenvs/digispark-tiny/lib379/DigisparkKeyboard/usbdrv.c.o
In file included from src/keyboard.cpp:1:0:
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '161' from 'int' to 'char' inside { } [-Wnarrowing]
};
^
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '224' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '231' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '149' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '129' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '149' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '129' from 'int' to 'char' inside { } [-Wnarrowing]
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h:59:1: error: narrowing conversion of '192' from 'int' to 'char' inside { } [-Wnarrowing]
*** [.pioenvs/digispark-tiny/src/keyboard.cpp.o] Error 1
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/osccal.c: In function 'calibrateOscillator':
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/osccal.c:34:13: warning: implicit declaration of function 'usbMeasureFrameLength' [-Wimplicit-function-declaration]
x = usbMeasureFrameLength();    /* proportional to current real frequency */
^
==================================================================== [ERROR] Took 1.12 seconds ====================================================================
```

Which looks like 3 issues:
1. redefinition of `BIN` with different values
2. deprecated inclusion of `delay.h` which should be `util/delay.h`
3. conversion of from 'int' to 'char' inside { }

tcurdt, 2018-11-05:
changing the file

```
~/.platformio/packages/framework-arduinoavr/libraries/__cores__/digispark/DigisparkKeyboard/DigiKeyboard.h
```

explicitly casting to `char` is a hack/workaround for at least error 3)

ekrempel, 2018-12-27:
Many thanks for your tip. I was able to compile and deploy after I changed to code like this:
`const PROGMEM char usbHidReportDescriptor[USB_CFG_HID_REPORT_DESCRIPTOR_LENGTH] = { /* USB report descriptor */
  (char) 0x05, (char) 0x01,                    // USAGE_PAGE (Generic Desktop)
  (char) 0x09, (char) 0x06,                    // USAGE (Keyboard)
  (char) 0xa1, (char) 0x01,                    // COLLECTION (Application)
  (char) 0x05, (char) 0x07,                    //   USAGE_PAGE (Keyboard)
  (char) 0x19, (char) 0xe0,                    //   USAGE_MINIMUM (Keyboard LeftControl)
  (char) 0x29, (char) 0xe7,                    //   USAGE_MAXIMUM (Keyboard Right GUI)
  (char) 0x15, (char) 0x00,                    //   LOGICAL_MINIMUM (0)
  (char) 0x25, (char) 0x01,                    //   LOGICAL_MAXIMUM (1)
  (char) 0x75, (char) 0x01,                    //   REPORT_SIZE (1)
  (char) 0x95, (char) 0x08,                    //   REPORT_COUNT (8)
  (char) 0x81, (char) 0x02,                    //   INPUT (Data,Var,Abs)
  (char) 0x95, (char) 0x01,           //   REPORT_COUNT (simultaneous keystrokes)
  (char) 0x75, (char) 0x08,                    //   REPORT_SIZE (8)
  (char) 0x25, (char) 0x65,                    //   LOGICAL_MAXIMUM (101)
  (char) 0x19, (char) 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
  (char) 0x29, (char) 0x65,                    //   USAGE_MAXIMUM (Keyboard Application)
  (char) 0x81, (char) 0x00,                    //   INPUT (Data,Ary,Abs)
  (char) 0xc0                           // END_COLLECTION
};`

ekrempel, 2018-12-27:
Many thanks for your tip. I was able to compile and deploy the coded after I modified the code as follows. 

```
const PROGMEM char usbHidReportDescriptor[USB_CFG_HID_REPORT_DESCRIPTOR_LENGTH] = { /* USB report descriptor */

  (char) 0x05, (char) 0x01,                    // USAGE_PAGE (Generic Desktop)
  (char) 0x09, (char) 0x06,                    // USAGE (Keyboard)
  (char) 0xa1, (char) 0x01,                    // COLLECTION (Application)
  (char) 0x05, (char) 0x07,                    //   USAGE_PAGE (Keyboard)
  (char) 0x19, (char) 0xe0,                    //   USAGE_MINIMUM (Keyboard LeftControl)
  (char) 0x29, (char) 0xe7,                    //   USAGE_MAXIMUM (Keyboard Right GUI)
  (char) 0x15, (char) 0x00,                    //   LOGICAL_MINIMUM (0)
  (char) 0x25, (char) 0x01,                    //   LOGICAL_MAXIMUM (1)
  (char) 0x75, (char) 0x01,                    //   REPORT_SIZE (1)
  (char) 0x95, (char) 0x08,                    //   REPORT_COUNT (8)
  (char) 0x81, (char) 0x02,                    //   INPUT (Data,Var,Abs)
  (char) 0x95, (char) 0x01,           //   REPORT_COUNT (simultaneous keystrokes)
  (char) 0x75, (char) 0x08,                    //   REPORT_SIZE (8)
  (char) 0x25, (char) 0x65,                    //   LOGICAL_MAXIMUM (101)
  (char) 0x19, (char) 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
  (char) 0x29, (char) 0x65,                    //   USAGE_MAXIMUM (Keyboard Application)
  (char) 0x81, (char) 0x00,                    //   INPUT (Data,Ary,Abs)
  (char) 0xc0                           // END_COLLECTION
};
```
```

maelremrem, 2019-03-12:
I have an error : "too many initializers for 'const char [22]'" on the same part of the lib, can you help me ?

Lukas1818, 2019-05-28:
I have create a pull request

steelywing, 2019-07-04:
I change to `unsigned char` also work.

```c
// .platformio\packages\framework-arduinoavr\libraries\__cores__\digispark\DigisparkKeyboard\DigiKeyboard.h

const PROGMEM unsigned char usbHidReportDescriptor[USB_CFG_HID_REPORT_DESCRIPTOR_LENGTH]
```

```c
// .platformio\packages\framework-arduinoavr\libraries\__cores__\digispark\DigisparkKeyboard\usbdrv.h

unsigned char usbDescriptorHidReport[];
```

bryanjhv, 2019-08-10:
I guess it's NOT a Digistump thing, because that code compiles fine on Arduino IDE.
That made me look at root cause, it's because `atmelavr` sets other flags for compilation.

Specifically, `builder/frameworks/arduino.py` should read:
```python
    CXXFLAGS=[
        "-Wno-error=narrowing",
```

kdlslyv, 2020-01-06:
just fyi, as of 2020-01-06, still had to apply the proposed fix by @bryanjhv manually to work with DigiSpark USB.

zaypen, 2020-08-13:
As of 2020-08-14, having the same issue 😂
alternative fix besides @bryanjhv 's, add this line to `platformio.ini`
```
build_flags = -Wno-error=narrowing
```

dromer, 2021-02-18:
Thnx @zaypen that did the trick for me!