platformio/platform-atmelavr #114: delay() function wrong timings aka. wrong fuses

Gurkengewuerz, 2018-11-06:
Tofay i tested the newest platformIO version with the ATtiny85. After testing the ATtiny with a blink sketech i recognized, that the delay() function is much longer than expected (1/2 of a second was about 4 seconds). This issue is related to #57.

I read out the fuses:
```
avrdude.exe: safemode: lfuse reads as 62
avrdude.exe: safemode: hfuse reads as DF
avrdude.exe: safemode: efuse reads as FF
avrdude.exe: safemode: Fuses OK (H:FF, E:DF, L:62)
```

After i set the fuses with avrdude to extended 0xFF, high 0xDF and low 0xE2 like in issue #57 it worked like a charm.
As described in the related issue: *[..] low fuse of 0x62 means that a clock divider by 8 is activated. The low fuse should be 0xe2, so that the device runs with 8 MHz.*

My platformio.ini is the following.
```
[common]
env_default = attiny85
upload_port = COM3
monitor_speed = 9600

[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
upload_protocol = stk500v1
upload_speed = 19200
upload_flags =
    -P$UPLOAD_PORT
    -b$UPLOAD_SPEED
monitor_speed = ${common.monitor_speed}
upload_port = ${common.upload_port}
```

strange-v, 2021-10-04:
I have the same issue using version 3.4.0.
After switching back to v3.2.0 everything works as expected.
```ini
[env:attiny85]
platform = atmelavr@3.2.0
board = attiny85
framework = arduino
board_build.f_cpu = 1000000L
board_hardware.oscillator = internal
board_fuses.lfuse = 0x62
board_fuses.hfuse = 0xDF
board_fuses.efuse = 0xFF
upload_protocol = usbasp
```

Basically, board_build.f_cpu is not used during the build, it always builds for 8000000L (or 4000000L, not sure).

linxcow, 2025-08-02:
I still have the same issue running an attiny85 at 1Mhz fuses set correctly
platform.ini set :
        board_build.f_cpu = 1000000L

        _delay_ms(3000);  // takes 24secs

any workaround for this?